evilsocket/pwnagotchi #981: SSH keys long generation

swanserquack, 2021-04-08:
## Expected Behaviour
The ssh keys should just generate normally.

## Current Behaviour
Stuck on the boot screen with it just trying to generate ssh keys.

## Context
I'm just trying to set up the pwnagotchi normally to use it for general testing purposes.

## Your Environment

- [Yes] You're using the official images

- [Yes] You're using a raspberry pi 0

- [no] You're using a supported LCD

The raspberry pi zero has been stuck generating ssh keys for the last 6 hours (at the time of writing). I have tried reflashing and nothing changes. I am using the instructions off https://pwnagotchi.ai/installation/.

aspman, 2021-04-17:
Have you installed the rndis driver for Windows 10? Ethernet gadget is no longer plug and play since a Windows update last year.

swanserquack, 2021-04-17:
> Have you installed the rndis driver for Windows 10? Ethernet gadget is no longer plug and play since a Windows update last year.

Sorry for the late response, I have now found the [correct driver](https://forum.moddevices.com/t/rndis-driver-for-windows-10/299/5) after searching for a few hours and following [these](https://community.pwnagotchi.ai/t/windows-10-connection-sharing-guide-thing-for-rpi0w-pwnagotchi-v-1-4-0-and-earlier/273) instructions to install the driver. I now have everything working but thanks for your concern.

closed 2021-04-17 by swanserquack