evilsocket/pwnagotchi #984: Pwnagotchi error not start/displaying ???

lildaze, 2021-04-09:
pi@pwnagotchi:/etc/systemd/system $ systemctl start pwnagotchi.service
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'pwnagotchi.service'.
Authenticating as: root
Password: 
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
==== AUTHENTICATION FAILED ===
Failed to start pwnagotchi.service: Access denied
See system logs and 'systemctl status pwnagotchi.service' for details.

pi@pwnagotchi:/etc/systemd/system $ systemctl status pwnagotchi.service
● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bett
   Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; vendor prese
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-07-10 03
     Docs: https://pwnagotchi.ai
  Process: 966 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAI
 Main PID: 966 (code=exited, status=1/FAILURE)

daberkow, 2021-04-17:
Have you tried "sudo systemctl start pwnagotchi.service"

evilsocket, 2021-04-24:
you should use sudo as @daberkow mentioned, the errors i see there are due to permissions, which is expected as you're running commands as normal user instead of root

closed 2021-04-25 by evilsocket