evilsocket/pwnagotchi #988: I can run Pwnagotchi on my raspberry Pi4B 8GB.

kangjiayin, 2021-04-11:
[image]

kangjiayin, 2021-04-11:
I can use all the system such as ubuntu raspberrypios without pwnagotchi.

lolzdatguy, 2021-04-11:
The image was not made for this board. Or at least that is my guess. You could install it manually, there is a guide for that on pwnagotchi.ai

kangjiayin, 2021-04-12:
> The image was not made for this board. Or at least that is my guess. You could install it manually, there is a guide for that on pwnagotchi.ai

Thanks, I will check my step again.

swanserquack, 2021-04-12:
> > The image was not made for this board. Or at least that is my guess. You could install it manually, there is a guide for that on pwnagotchi.ai
> 
> Thanks, I will check my step again.

I got the same issue. Followed all the steps. I think its a firmware issue so its out of our hands.

kangjiayin, 2021-04-20:
> > > The image was not made for this board. Or at least that is my guess. You could install it manually, there is a guide for that on pwnagotchi.ai
> > 
> > 
> > Thanks, I will check my step again.
> 
> I got the same issue. Followed all the steps. I think it's a firmware issue so it is out of our hands.

Have you ever try to install it on Linux?

> > > The image was not made for this board. Or at least that is my guess. You could install it manually, there is a guide for that on pwnagotchi.ai
> > 
> > 
> > Thanks, I will check my step again.
> 
> I got the same issue. Followed all the steps. I think its a firmware issue so its out of our hands.

Have you ever try to install it on Linux?

kangjiayin, 2021-04-20:
> > > The image was not made for this board. Or at least that is my guess. You could install it manually, there is a guide for that on pwnagotchi.ai
> > 
> > 
> > Thanks, I will check my step again.
> 
> I got the same issue. Followed all the steps. I think its a firmware issue so its out of our hands.

Have you ever try to install it on Linux?

swanserquack, 2021-04-20:
> Have you ever try to install it on Linux?

What do you mean by that?

donut-donut, 2021-04-22:
works on a rpi 4 4gb just fine

swanserquack, 2021-04-22:
> works on a rpi 4 4gb just fine

It's the RAM chip that's the problem. It happens when trying to install kali Linux as well. It's because the 8GB model has a different RAM chip and so it need new firmware and other things to make it work.

kangjiayin, 2021-04-22:
> > works on a rpi 4 4gb just fine
> 
> It's the RAM chip that's the problem. It happens when trying to install kali Linux as well. It's because the 8GB model has a different RAM chip and so it need new firmware and other things to make it work.

Thx

evilsocket, 2021-04-24:
the image is made for raspberry pi zero w boards, not this

evilsocket, 2021-04-24:
https://pwnagotchi.ai/installation/#installing-on-any-gnu-linux

closed 2021-04-24 by evilsocket

kangjiayin, 2021-04-24:
> the image is made for raspberry pi zero w boards, not this

[image]

evilsocket, 2021-04-24:
right, should. 

it's not officially built for Rpi3 and Rpi3 models and it should work on some of them, i don't test every hardware revision of other boards other than pi0w and it's likely that compatibility can break for newer ones. That is why I linked Linux generic instructions so that one can work with raspbian or whatever distro they have running on whatever board.

kangjiayin, 2021-04-24:
> right, should.
>
> it's not officially built for Rpi3 and Rpi3 models and it should work on some of them, i don't test every hardware revision of other boards other than pi0w and it's likely that [...]

Thank you.
best wishes.