evilsocket/pwnagotchi #992: AI not loaded on fresh installation (v 1.5.5)

BarnOwlDown, 2021-04-20:
Raspberry Pi Zero W, waveshare v2, fresh installation of Pwnagotchi 1.5.5.

I start Pwnagotchi with this config.toml in boot partition:

```
main.name = "pwnagotchi"
main.lang = "en"
main.whitelist = [
  "mynetwork",
  "00:00:00:00:00:00"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "mynetwork",
  "00:00:00:00:00:00"
]

ui.display.enabled = true
ui.display.type = "waveshare_2"
ui.display.color = "black"

main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.pixel.enabled = true
main.plugins.bt-tether.devices.pixel.search_order = 1
main.plugins.bt-tether.devices.pixel.mac = "00:00:00:00:00:00"
main.plugins.bt-tether.devices.pixel.ip = "192.168.44.88"
main.plugins.bt-tether.devices.pixel.netmask = 24
main.plugins.bt-tether.devices.pixel.interval = 1
main.plugins.bt-tether.devices.pixel.scantime = 15
main.plugins.bt-tether.devices.pixel.share_internet = true
main.plugins.bt-tether.devices.pixel.priority = 1
main.plugins.bt-tether.devices.pixel.max_tries = 10
```

and I have this error in pwnagotchi.log

```
[2021-04-20 10:06:56,604] [ERROR] error while starting AI (numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/__init__.py", line 21, in load
    from stable_baselines import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/__init__.py", line 1, in <module>
    from stable_baselines.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/__init__.py", line 1, in <module>
    from stable_baselines.a2c.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 6, in <module>
    import tensorflow as tf
  File "/usr/lib/python3/dist-packages/tensorflow/__init__.py", line 98, in <module>
    from tensorflow_core import *
  File "/usr/lib/python3/dist-packages/tensorflow_core/__init__.py", line 35, in <module>
    from tensorflow._api.v1 import compat
  File "/usr/lib/python3/dist-packages/tensorflow_core/_api/v1/compat/__init__.py", line 21, in <module>
    from tensorflow._api.v1.compat import v1
  File "/usr/lib/python3/dist-packages/tensorflow_core/_api/v1/compat/v1/__init__.py", line 652, in <module>
    from tensorflow_estimator.python.estimator.api._v1 import estimator
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/__init__.py", line 8, in <module>
    from tensorflow_estimator._api.v1 import estimator
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/_api/v1/estimator/__init__.py", line 8, in <module>
    from tensorflow_estimator._api.v1.estimator import experimental
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/_api/v1/estimator/experimental/__init__.py", line 8, in <module>
    from tensorflow_estimator.python.estimator.canned.dnn import dnn_logit_fn_builder
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/__init__.py", line 25, in <module>
    import tensorflow_estimator.python.estimator.estimator_lib
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/estimator_lib.py", line 53, in <module>
    from tensorflow_estimator.python.estimator.inputs import inputs
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/inputs.py", line 22, in <module>
    from tensorflow_estimator.python.estimator.inputs.numpy_io import numpy_input_fn
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/numpy_io.py", line 26, in <module>
    from tensorflow_estimator.python.estimator.inputs.queues import feeding_functions
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/queues/feeding_functions.py", line 40, in <module>
    import pandas as pd
  File "/usr/local/lib/python3.7/dist-packages/pandas/__init__.py", line 29, in <module>
    from pandas._libs import hashtable as _hashtable, lib as _lib, tslib as _tslib
  File "/usr/local/lib/python3.7/dist-packages/pandas/_libs/__init__.py", line 13, in <module>
    from pandas._libs.interval import Interval
  File "pandas/_libs/interval.pyx", line 1, in init pandas._libs.interval
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject
[2021-04-20 10:06:56,704] [WARNING] [ai] AI not loaded!
```

noddynz, 2021-04-21:
I get this too on a fresh install

mzielinskim, 2021-04-21:
I got that message either after fresh install... reverted to 1.5.3

wedemmoez, 2021-04-21:
Same issue here

infinition, 2021-04-22:
same here too

infinition, 2021-04-22:
> same here too

Solved with a "sudo3 pip install --upgrade numpy"

hevv88, 2021-04-22:
> > same here too
> 
> Solved with a "sudo3 pip install --upgrade numpy"

Is the command corect ?

nathomp, 2021-04-22:
> > > same here too
> > 
> > 
> > Solved with a "sudo3 pip install --upgrade numpy"
> 
> Is the command corect ?

sudo pip3*

infinition, 2021-04-22:
> > > > same here too
> > > 
> > > 
> > > Solved with a "sudo3 pip install --upgrade numpy"
> > 
> > 
> > Is the command corect ?
> 
> sudo pip3*

Yeah sorry , it was 4am lool.
"sudo pip3 install --upgrade numpy"

nathomp, 2021-04-22:
> > > > > same here too
> > > > 
> > > > 
> > > > Solved with a "sudo3 pip install --upgrade numpy"
> > > 
> > > 
> > > Is the command corect ?
> > 
> > 
> > sudo pip3*
> 
> Yeah sorry , it was 4am lool.
> "sudo pip3 install --upgrade numpy"

Gives a warning about incompatible versions but still works.

aspman, 2021-04-22:
Is there an easier/alternative way to upgrade numpy than this when your only reliable internet connection is via phone Bluetooth which keeps timing out? 😬 it's not the same point in the download it fails at and it's getting annoying 😅

infinition, 2021-04-22:
> Is there an easier/alternative way to upgrade numpy than this when your only reliable internet connection is via phone Bluetooth which keeps timing out? 😬 it's not the same point in the download it fails at and it's getting annoying 😅

Try to add this  : 

sudo vi /etc/resolv.conf
	nameserver 127.0.0.1
	nameserver 8.8.8.8

infinition, 2021-04-22:
> > > > > > same here too
> > > > > 
> > > > > 
> > > > > Solved with a "sudo3 pip install --upgrade numpy"
> > > > 
> > > > 
> > > > Is the command corect ?
> > > 
> > > 
> > > sudo pip3*
> > 
> > 
> > Yeah sorry , it was 4am lool.
> > "sudo pip3 install --upgrade numpy"
> 
> Gives a warning about incompatible versions but still works.

Yes, for now, everything is working fine.

ljorge705, 2021-04-22:
> > > > > same here too
> > > > 
> > > > 
> > > > Solved with a "sudo3 pip install --upgrade numpy"
> > > 
> > > 
> > > Is the command corect ?
> > 
> > 
> > sudo pip3*
> 
> Yeah sorry , it was 4am lool.
> "sudo pip3 install --upgrade numpy"

I give this error when update numpy

`[2021-04-22 19:31:01,803] [ERROR] error while starting AI (

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/usr/bin/python3"
  * The NumPy version is: "1.20.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: /usr/local/lib/python3.7/dist-packages/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so: file too short
)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: /usr/local/lib/python3.7/dist-packages/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so: file too short

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/__init__.py", line 21, in load
    from stable_baselines import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/__init__.py", line 1, in <module>
    from stable_baselines.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/__init__.py", line 1, in <module>
    from stable_baselines.a2c.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 4, in <module>
    import gym
  File "/usr/local/lib/python3.7/dist-packages/gym/__init__.py", line 10, in <module>
    from gym.spaces import Space
  File "/usr/local/lib/python3.7/dist-packages/gym/spaces/__init__.py", line 1, in <module>
    from gym.spaces.space import Space
  File "/usr/local/lib/python3.7/dist-packages/gym/spaces/space.py", line 1, in <module>
    from gym.utils import seeding
  File "/usr/local/lib/python3.7/dist-packages/gym/utils/seeding.py", line 2, in <module>
    import numpy as np
  File "/usr/local/lib/python3.7/dist-packages/numpy/__init__.py", line 145, in <module>
    from . import core
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/usr/bin/python3"
  * The NumPy version is: "1.20.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: /usr/local/lib/python3.7/dist-packages/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so: file too short

[2021-04-22 19:31:02,102] [WARNING] [ai] AI not loaded!`

SammCheese, 2021-04-22:
One way to upgrade and still have AI running is by flashing the 1.5.3 Image and using auto update to update to 1.5.5

ljorge705, 2021-04-23:
> One way to upgrade and still have AI running is by flashing the 1.5.3 Image and using auto update to update to 1.5.5

Works, thanks for help

adi170-alt, 2021-04-23:
I hope that evilsocket fixes this soon because I have now a normal good working pwangotchi and a pwangotochi without ai lol

BarnOwlDown, 2021-04-23:
Is it better to do "sudo apt update && sudo apt dist-upgrade" after versione upgrade from 1.5.3 to 1.5.5?

Btw, upgrade pwnagotchi from 1.5.3 works!

adi170-alt, 2021-04-24:
> Is it better to do "sudo apt update && sudo apt dist-upgrade" after versione upgrade from 1.5.3 to 1.5.5?
> 
> Btw, upgrade pwnagotchi from 1.5.3 works!

yea, I'm going to do that, but I asked evilsocket when he is going to fix it, And he said this: https://twitter.com/evilsocket/status/1385675286700871680

adi170-alt, 2021-04-24:
> Is it better to do "sudo apt update && sudo apt dist-upgrade" after versione upgrade from 1.5.3 to 1.5.5?
> 
> Btw, upgrade pwnagotchi from 1.5.3 works!

I get this when i run that command :(

```
pi@DEKU:~ $ sudo apt update && sudo apt dist-upgrade
Err:1 http://archive.raspberrypi.org/debian buster InRelease
  Temporary failure resolving 'archive.raspberrypi.org'
Err:2 http://http.re4son-kernel.com/re4son kali-pi InRelease
  Temporary failure resolving 'http.re4son-kernel.com'
Err:3 http://raspbian.raspberrypi.org/raspbian buster InRelease
  Temporary failure resolving 'raspbian.raspberrypi.org'
Reading package lists... Done
Building dependency tree
Reading state information... Done
5 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Temporary failure resolving 'raspbian.raspberrypi.org'
W: Failed to fetch http://http.re4son-kernel.com/re4son/dists/kali-pi/InRelease  Temporary failure resolving 'http.re4son-kernel.com'
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/buster/InRelease  Temporary failure resolving 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree
  firmware-realtek
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
```

BarnOwlDown, 2021-04-24:
> > Is it better to do "sudo apt update && sudo apt dist-upgrade" after versione upgrade from 1.5.3 to 1.5.5?
> > Btw, upgrade pwnagotchi from 1.5.3 works!
> 
> I get this when i run that command :(
> 
> ```
> pi@DEKU:~ $ sudo apt update && sudo apt dist-upgrade
> Err:1 http://archive.raspberrypi.org/debian buster InRelease
>   Temporary failure resolving 'archive.raspberrypi.org'
> Err:2 http://http.re4son-kernel.com/re4son kali-pi InRelease
>   Temporary failure resolving 'http.re4son-kernel.com'
> Err:3 http://raspbian.raspberrypi.org/raspbian buster InRelease
>   Temporary failure resolving 'raspbian.raspberrypi.org'
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> 5 packages can be upgraded. Run 'apt list --upgradable' to see them.
> W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Temporary failure resolving 'raspbian.raspberrypi.org'
> W: Failed to fetch http://http.re4son-kernel.com/re4son/dists/kali-pi/InRelease  Temporary failure resolving 'http.re4son-kernel.com'
> W: Failed to fetch http://archive.raspberrypi.org/debian/dists/buster/InRelease  Temporary failure resolving 'archive.raspberrypi.org'
> W: Some index files failed to download. They have been ignored, or old ones used instead.
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Calculating upgrade... Done
> The following packages have been kept back:
>   firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree
>   firmware-realtek
> 0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
> ```
> 
> and yes, My pwangotchi is connected to the internet
> 
> and somehow autoupdate does not work because it does not update

This is your DNS problem. Check resolv.conf or internet connection with "ping 1.1.1.1".

adi170-alt, 2021-04-24:
> This is your DNS problem. Check resolv.conf or internet connection with "ping 1.1.1.1".

OHH, You are right!, a few days ago I installed pi-hole on my raspberry pi 4. I'm gonna try fixing it on pi hole lol, thanks

infinition, 2021-04-24:
> OHH, You are right!, a few days ago I installed pi-hole on my raspberry pi 4. I'm gonna try fixing it on pi hole lol, thanks
This should solve your issue ;

sudo vi /etc/resolv.conf
nameserver 127.0.0.1
nameserver 8.8.8.8

adi170-alt, 2021-04-24:
> This should solve your issue ;
> sudo vi /etc/resolv.conf
> nameserver 127.0.0.1
> nameserver 8.8.8.8

thank u lol, that worked

evilsocket, 2021-04-24:
sorry guys everything went smooth while i upgraded from 1.5.3 to 1.5.5 so I assumed the fresh install would have worked as well. 

Can some of the guys who fixed this issue by upgrading numpy run this command on their unit (that should be a fresh install of 1.5.5 fixed with that upgrade) and paste here the output please:

sudo pip3 show numpy

evilsocket, 2021-04-24:
fixed the DNS server issue in the meantime

evilsocket, 2021-04-24:
the funny thing is that both my older unit and a freshly installed 1.5.5 have numpy==1.17.2 ... this bug doesn't make a lot of sense but will try to fix it anyway ...

closed 2021-04-24 by evilsocket

adi170-alt, 2021-04-24:
How to update my pwnagotchi with the fixes? lol

evilsocket, 2021-04-24:
as previously wrote:

> sudo pip3 install --upgrade numpy

(you don't have to end every sentence with lol you know? lol)

adi170-alt, 2021-04-24:
> as previously wrote:
> 
> > sudo pip3 install --upgrade numpy
> 
> (you don't have to end every sentence with lol you know? lol)

oh sorry, you are right

evilsocket, 2021-04-24:
thanks for making me focus on this issue, i totally overlooked it :D lol <3

infinition, 2021-04-24:
> sorry guys everything went smooth while i upgraded from 1.5.3 to 1.5.5 so I assumed the fresh install would have worked as well.
>
> Can some of the guys who fixed this issue by upgrading numpy run [...]

Hi mate, sorry for the late reply.
Strangely, the pip3 show numpy command freeze and nothing happens... i'll have a look on that too.

I ran the same command to show you the installed version of numpy, stable with 1.5.5 (as far as we have seen).

root@*********:~# sudo pip3 install --upgrade numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: numpy in /usr/local/lib/python3.7/dist-packages (1.20.2)

Have a great day !

TheREK3R, 2021-05-02:
Has this been fixed in the newest release? I see this is closed but I just installed the newest version onto a new rpi0 and it isn't switching into ai mode either.

SammCheese, 2021-05-02:
> Has this been fixed in the newest release? I see this is closed but I just installed the newest version onto a new rpi0 and it isn't switching into ai mode either.

Try this

> One way to upgrade and still have AI running is by flashing the 1.5.3 Image and using auto update to update to 1.5.5

TheREK3R, 2021-05-02:
I tried this last night but it bricked itself during the update and I have to re-flash again, which I am doing now

TheREK3R, 2021-05-02:
yeah, it keeps bricking itself on update, not sure why. I might just stay on 1.5.5 with no ai for now

aspman, 2021-05-02:
> yeah, it keeps bricking itself on update, not sure why. I might just stay on 1.5.5 with no ai for now

When you say it bricks, do you mean full on dead to rights, or just stuck even if left a while? I left mine to update a fresh 1.5.3 over night and it seemed to go ok. (My internet was over BT tether to my phone, so it takes a while doing anything, sometimes even timing out and needs running again)

TheREK3R, 2021-05-03:
my understanding is that running apt update is unnecessary and the only reqs for auto upgrade are internet access and time, is that correct? The plugins page for autoupdate says its a wrapper for apt update + upgrade so I figured that its fine to run apt update on my own.

TheREK3R, 2021-05-03:
I just had an idea, what if my computer isnt supplying enough power and it crashes the pi during upgrading something critical? I am going to find another cord or plug in powersupply into the power as well to see if this is the case.

TheREK3R, 2021-05-03:
Thats not it :( it keeps dying to the point it wont flash the led every time it updates in 1.53 or 1.55. Its the standard configuration (only changed the hostname) with pi0-WH and waveshare 2 so IDK whats up with it. I think maybe its an upstream package issue that breaks something on apt upgrade, so if someone else is inclined they can test as well that could be good. I think I will install 1.5.3 and not update it for now. I will get ai mode at least.

TheREK3R, 2021-05-03:
okay after the pi bricks itself I get this in the boot partition. I think something is corrupted here. I am going to try replacing the sd card, but after that I will install raspbian lite and see if maybe the pi itself is just defunct.
[image]

scifijunk, 2022-02-05:
Well to fix the AI Mode problem I used the fix that @dadav had in his pwnagotchi/ai/__init__.py

by changing 
```
            a2c.load(config['path'], env)
            logging.debug("[ai] A2C loaded in %.2fs" % (time.time() - start))
```

to
```
            try:
                a2c.load(config['path'], env)
            except AssertionError as as_err:
                from fnmatch import fnmatch
                # Sometimes the model breaks...
                if not fnmatch(str(as_err), '* same * space as the model *'):
                    raise as_err
                else:
                    logging.warning("[ai] Model could not be loaded. Using new model.")
            else:
                logging.debug("[ai] A2C loaded in %.2fs" % (time.time() - start))
```

which is in my PR titled The quest to make AI mode work #1066. Mind you I'm a Github noob and was unaware that the changed I made after the PR would be reflected until @fireneat stated "You forgot to change this back". I even went as far as to put together a Gitea git server so I may continue to work on my fork of pwnagotchi without it reflecting my changes in my PR and just in case Microsoft starts to pull projects from Github.

XZCendence, 2022-02-27:
> I got that message either after fresh install... reverted to 1.5.3

How exactly do you revert your pwnagotchi version?

noddynz, 2022-04-17:
For me personally I just backed up my config's and just re-imaged.

Spessmen14, 2022-10-19:
I'm having a monsterous time getting the damn tethering to work so i can reinstall numpy. Is there a way to download the correct version, put it on a microSD card, and install it from local files?

kevz0r, 2022-12-09:
OMG i finally my AI working after lots and lots of trial and error and TBH I'm not too sure what fixed it. no brain.nn being created.

i had mainly been coming up against this error, which a lot of people struggled with:

```
[ERROR] [ai] error while training (could not broadcast input array from shape (428,) into shape (1,503))
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/home/pi/stable-baselines/stable_baselines/a2c/a2c.py", line 267, in learn
    rollout = self.runner.run(callback)
  File "/home/pi/stable-baselines/stable_baselines/common/base_class.py", line 794, in runner
    self._runner = self._make_runner()
  File "/home/pi/stable-baselines/stable_baselines/a2c/a2c.py", line 110, in _make_runner
    return A2CRunner(self.env, self, n_steps=self.n_steps, gamma=self.gamma)
  File "/home/pi/stable-baselines/stable_baselines/a2c/a2c.py", line 342, in __init__
    super(A2CRunner, self).__init__(env=env, model=model, n_steps=n_steps)
  File "/home/pi/stable-baselines/stable_baselines/common/runners.py", line 31, in __init__
    self.obs[:] = env.reset()
  File "/home/pi/stable-baselines/stable_baselines/common/vec_env/dummy_vec_env.py", line 62, in reset
    self._save_obs(env_idx, obs)
  File "/home/pi/stable-baselines/stable_baselines/common/vec_env/dummy_vec_env.py", line 92, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (428,) into shape (1,503)
```

the few things that I had done recently are below - hopefully it can help someone. I'm running a pizero2 with a 5ghz USB dongle:

- made this change: https://github.com/evilsocket/pwnagotchi/issues/992#issuecomment-1030504516
- changed the line in `/etc/modprobe.d/8821au.conf` to `options 8821au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_dfs_region_domain=0 rtw_RFE_type=7 rtw_country_code=US`
- tried a bunch of random pip combinations - package list below

hope this helps someone!

```
absl-py==1.3.0
ansible==4.10.0
ansible-core==2.11.12
apache-libcloud==3.6.1
asn1crypto==1.5.1
astor==0.8.1
atari-py==0.3.0
bcrypt==3.1.6
cached-property==1.5.2
cachetools==5.2.0
certifi==2022.12.7
cffi==1.15.1
chardet==5.1.0
charset-normalizer==2.1.1
click==8.1.3
cloudpickle==1.2.2
colorzero==2.0
cryptography==2.6.1
cycler==0.11.0
Cython==0.29.32
dbus-python==1.3.2
distro==1.8.0
entrypoints==0.4
file-read-backwards==2.0.0
Flask==2.2.2
Flask-Cors==3.0.10
Flask-WTF==1.0.1
fonttools==4.38.0
future==0.18.2
gast==0.5.3
google-auth==2.15.0
google-auth-oauthlib==0.7.1
google-pasta==0.2.0
gpiozero==1.6.2
grpcio==1.51.1
gym==0.15.7
gym-notices==0.0.8
h5py==3.7.0
httplib2==0.21.0
idna==3.4
importlib-metadata==5.1.0
inky==1.4.0
install==1.3.5
itsdangerous==2.1.2
jaraco.classes==3.2.3
jeepney==0.8.0
Jinja2==3.1.2
jmespath==1.0.1
joblib==1.2.0
Keras-Applications==1.0.8
Keras-Preprocessing==1.1.2
keyring==23.11.0
keyrings.alt==4.2.0
kiwisolver==1.4.4
lockfile==0.12.2
Markdown==3.4.1
MarkupSafe==2.1.1
matplotlib==3.5.3
more-itertools==9.0.0
mpi4py==3.1.4
netaddr==0.8.0
ntlm-auth==1.5.0
numpy==1.21.6
oauthlib==3.2.2
olefile==0.46
opencv-python==4.6.0.66
opt-einsum==3.3.0
packaging==22.0
pandas==1.3.5
paramiko==2.12.0
Pillow==9.3.0
plotly==5.11.0
protobuf==4.21.11
pwnagotchi==1.5.5
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycairo==1.23.0
pycparser==2.21
pycrypto==2.6.1
pycryptodome==3.16.0
pyglet==1.5.0
PyGObject==3.30.4
pykerberos==1.1.14
PyNaCl==1.5.0
pyparsing==3.0.9
PySocks==1.7.1
python-apt==1.8.4.3
python-dateutil==2.8.2
pytz==2022.6
pywinrm==0.4.3
pyxdg==0.28
PyYAML==6.0
requests==2.28.1
requests-kerberos==0.11.0
requests-ntlm==1.1.0
requests-oauthlib==1.3.1
resolvelib==0.5.5
RPi.GPIO==0.7.1
rsa==4.9
scapy==2.4.5
scipy==1.7.3
SecretStorage==3.3.3
simplejson==3.18.0
six==1.16.0
smbus2==0.4.2
spidev==3.6
ssh-import-id==5.11
-e git+https://github.com/hill-a/stable-baselines.git@45beb246833b6818e0f3fc1f44336b1c52351170#egg=stable_baselines
tenacity==8.1.0
tensorboard==1.13.1
tensorboard-data-server==0.6.1
tensorboard-plugin-wit==1.8.1
tensorflow==1.13.1
tensorflow-estimator==1.14.0
termcolor==2.1.1
toml==0.10.2
tweepy==4.12.1
typing-extensions==4.4.0
urllib3==1.26.13
websockets==10.4
Werkzeug==2.2.2
wrapt==1.14.1
WTForms==3.0.1
xmltodict==0.13.0
zipp==3.11.0
```

XZCendence, 2022-12-09:
>    - made this change: #992 (comment)
> -e ***@***.***#egg=stable_baselines
> zipp==3.11.0 ***@***.***#egg=stable_baselinestenacity==8.1.0tensorboard==1.13.1tensorboard-data-server==0.6.1tensorboard-plugin-wit==1.8.1tensorflow==1.13.1tensorflow-estimator==1.14.0termcolor==2.1.1toml==0.10.2tweepy==4.12.1typing-extensions==4.4.0urllib3==1.26.13websockets==10.4Werkzeug==2.2.2wrapt==1.14.1WTForms==3.0.1xmltodict==0.13.0zipp==3.11.0>

I got this error far too many times and eventually gave up, I can't thank
you enough, I'll have to try this soon.

Delearner, 2023-11-22:
> > > > > same here too
> > > > 
> > > > 
> > > > Solved with a "sudo3 pip install --upgrade numpy"
> > > 
> > > 
> > > Is the command corect ?
> > 
> > 
> > sudo pip3*
> 
> Yeah sorry , it was 4am lool. "sudo pip3 install --upgrade numpy"

Every time i try to run it i get this error:
pi@Pwner:~ $ sudo pip3 install --upgrade numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5543210>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5543db0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5543dd0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5543a70>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb577ec10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5324350>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5324cb0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5324fb0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb53249b0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5324970>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/numpy/
Requirement already up-to-date: numpy in /usr/local/lib/python3.7/dist-packages (1.17.2)

project-draven, 2023-11-25:
> Every time i try to run it i get this error: pi@Pwner:~ $ sudo pip3 install --upgrade numpy Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple Retrying (Retry(total=4, [...]

Everytime you reboot your pwnagotchi the settings in the resolv.conf file rewrites to 127.0.0.1
sudo nano /etc/resolv.conf
then add nameserver 8.8.8.8 
ctrl+x then Y to save 
then try to update.

lucidyan, 2024-01-03:
@kevz0r Thanks for the insight! It seems that the problem with 
```
ValueError: could not broadcast input array from shape (428,) into shape (1,503)
```

Solved with the installing master version of stable-baselines (unfortunately unmaintainable at this time)
```
pip3 install --upgrade git+https://github.com/hill-a/stable-baselines.git@45beb246833b6818e0f3fc1f44336b1c52351170#egg=stable_baselines
```

Also could recommend [live fork of pwnagotchi](https://github.com/aluminum-ice/pwnagotchi) for the new strangers here

ljorge705, 2024-01-04:
> @kevz0r Thanks for the insight! It seems that the problem with
>
> ```
> ValueError: could not broadcast input array from shape (428,) into shape (1,503)
> ```
>
> Solved with the installing master version of [...]

Or [THIS](https://github.com/jayofelony/pwnagotchi-torch)

commit 20d80bfb (2021-04-24): fix: upgraded numpy version (fixes #992)