evilsocket/pwnagotchi #996: [BUG] No Connection to the Internet / Network

Luca-Hiller, 2021-05-05:
I can't connect to the internet with my pwnagotchi, but I can connect to ssh and use the 10.0.0.2 web interface

**Expected behavior**
No Connection to the internet

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5
 - Type of hardware Win 10 Pro / USB Data Cable

[image]
[image]

kevintully, 2021-05-06:
Have you run the PowerShell scripts to share your network connection with your pwnagotchi yet? (see: https://pwnagotchi.ai/configuration/#host-connection-sharing)

If so, you may want to check what nameservers are in use in the pwnagotchi with `cat /etc/resolv.conf`. If you see something like `nameserver 127.0.0.1` then you may need to edit dnsmasq settings to ensure that a resolvable DNS server is set at boot time. One way to do this is to add the following line to the bottom of /etc/dnsmasq.conf  `server=1.1.1.1@usb0`.

Luca-Hiller, 2021-05-07:
Many thanks for your help. Unfortunately, this didn't help. The nameservers are set to `8.8.8.8` for me. I added the line `server=1.1.1.1@usb0`, but the nameserver stays at 8.8.8.8. The problem remains

Nameserver
[image]

lolzdatguy, 2021-06-03:
There are so many reasons for ICS not to work. I suggest you just use Bluetooth tethering.

0x0lli, 2021-09-12:
neither the connection sharing nor the BT tethering works for me. I am connected in both cases but with ICS, updating or installing dont work, with BT, its connected but I have no idea how to connect to the shell there. I set everything on true in config.toml, the MAC of my phone's BT and a suggested IP.
how does one connect to it then?

DarthKegRaider, 2022-02-07:
0x0lli :  I had similar problems with BT.  It took me HOURS of searching the internet, trying to get connected.  I ended up using another phone to connect to my phone's BT tether, so I could find the gateway IP which for me was 192.168.44.1
Also, you need to change the first line from false to be true.

`main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = "00:11:22:AA:BB:CC"
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.2"`

After this, it sprang to life and on my normal phone running as the BT hotspot tether, I entered 192.168.44.2:8080 in the browser, was asked for my user:password, and working!

TenThousand1, 2023-08-22:
Darth, are you on android or iOS?

DarthKegRaider, 2023-08-24:
I'm using Android.  On a Samsung S22Ultra.

So what I meant about the 'other phone', was setup my S22Ultra as the Bluetooth tether with the internet connection hotspot.
Connected with my older Galaxy S8 to obtain the Gateway details to then enter into my Tamagotchi's configuration file.

Hope this helps.

mstDenim, 2024-02-14:
@Luca-Hiller 
Thank you, this helped a lot!

xiaoyao9184, 2024-06-21:
Make sure the firewall is functioning properly

```pwsh
$networkProfiles = Get-NetConnectionProfile
$networkProfiles | Format-Table -Property Name, InterfaceAlias, NetworkCategory
```

Change NetworkCategory to `Private`

```pwsh
Set-NetConnectionProfile -InterfaceAlias '<pwnagotchi>' -NetworkCategory Private
```