evilsocket/pwnagotchi #997: [BUG] Does not go into AI mode V1.5.5 (log included)

CynicalMexican, 2021-05-05:
Once the pwnagotchi goes into AUTO mode it boots up and freezes at "Hack The World!" or smth like that. Yes, I even reflashed
my SD card and reconfigured everything.

How I got it to do that (possibly)
1. Enabling bt-tethering
2. Enabling web-ui & webcfg
3. Changing addresses of web-ui and bettercap to "192.168.44.44" and "192.168.44.45"

What I was hoping it would happen
Able to use bt-tethering with my device and web-ui (I got it to work once but after that, the issues began)

**Screenshots**
Log: 
[pwnagotchi.log](https://github.com/evilsocket/pwnagotchi/files/6430473/pwnagotchi.log)

My config.toml (had to convert to .txt for uploading purposes also censored some personal information with X's):
[config.txt](https://github.com/evilsocket/pwnagotchi/files/6430482/config.txt)

My bnep-cfg file:
[bnep0-cfg.txt](https://github.com/evilsocket/pwnagotchi/files/6430492/bnep0-cfg.txt)

A picture of the pwnagotchi in this phase:
[image]

**Environment**
 - Pwnagotchi version : v1.5.5 
 - OS version : Windows 10
 - Type of hardware : Raspberry Pi Zero W

aspman, 2021-05-05:
I'm going to hazard a guess as bettercap can't connect to the API that the change in IP is conflicting with the config. It has its own configurations which could need updating to reflect your alterations. But that's an off-the-head guess at present as I'm still learning the ins and outs of it all. I recommend not changing IP from default and seeing if that rectifies the issue.

Also, I noticed that the log shows numpy errors, which affects ai from loading. You'll either need to solve the above (fresh flash, why not) and then update numpy as noted in a recent batch off bug submissions, or alternatively, flash 1.5.3 and allow it to update to 1.5.5 (which doesn't require updating numpy).

See how that goes and let us know.

CynicalMexican, 2021-05-05:
> I'm going to hazard a guess as bettercap can't connect to the API that the change in IP is conflicting with the config. It has its own configurations which could need updating to reflect your [...]

I will try that when I get the chance (cant find a data cable anywhere, its all power only :( )

CynicalMexican, 2021-05-07:
> I'm going to hazard a guess as bettercap can't connect to the API that the change in IP is conflicting with the config. It has its own configurations which could need updating to reflect your [...]

It works the AI is now being loaded, however new issue arises. Since it updated itself to the latest version (updated from 1.5.3), I've been having issues with bt-tethering and web-ui. Typing "192.168.44.44:8080" doesn't load at all but typing "192.168.44.44" loads up bettercap, but trying to log in doesn't do anything. I've tried countless hours (seriously) editing all possible combination of "bettercap.hostname" and ui.web.address" and nothing. I even tried to go back to the default settings of config.toml and still nothing. Most of the time I look at the logs and says "Waiting for bettercap API to be available" (I also made sure all credentials matched on all essential files for bettercap). On default no issues with bettercap but now the web-ui for my phone doesn't work. The only thing that I managed to get farthest in is going into MANU mode and restarting the pwnagotchi into AUTO mode from there. Should I go back to V1.5.3 and turn off auto-update, maybe wait till a new update comes?

CynicalMexican, 2021-05-07:
I'm starting to feel that web.ui isn't supposed to be able to be seen on a phone when the pwnagotchi is on AUTO/AI mode. Bettercap maybe... Welp, if that's the case then I hate myself (although I did learn a LOT of how to work a pwnagotchi by trying to troubleshoot for almost 3 days straight). I also learned almost everything by memory. Additionally, it seems the main purpose of bt-tethering a pwnagotchi is to give it only internet connection and not the ability to use web.ui. (That would be the best though) I'll close this post tomorrow incase someone has a detailed reason to why this is set up this way.

evilsocket, 2021-05-07:
duplicate of https://github.com/evilsocket/pwnagotchi/issues/992

closed 2021-05-07 by evilsocket