evilsocket/pwnagotchi #999: [BUG] Bettercap not displaying any captures/events

marko-kovacevic, 2021-05-11:
**Describe the bug**
A clear and concise description of what the bug is.

After collecting new networks with my Pi, I change back to manual mode, and log into bettercap. I don't see any new networks that I've gained, nor the old ones.

**To Reproduce**
Steps to reproduce the behavior:
1. ... Capture networks in AUTO Mode
2. ... Switch to manual mode
3. ... Visit ip address of Pi
4. Log into Bettercap
5. No networks or activity shown

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version
 - OS version
 - Type of hardware
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

evilsocket, 2021-05-28:
bettercap's ui is made to see realtime data, not historical data

marko-kovacevic, 2021-05-28:
> bettercap's ui is made to see realtime data, not historical data

That makes sense given the device, however I should still see some events/networks shown, no?

evilsocket, 2021-05-28:
not if you don't enable the modules ... you're in MANU mode, things are not enabled ... click on the wifi icon and enable the module