evilsocket/pwnagotchi #1002: Linux Connection Sharing does not work

xSetBack, 2021-05-26:
I have downloaded the bash script directly from the repo, i have changed the interfaces names properly, but i get every time the same error:

Temporary failure resolving '[url]'.

My host laptop: Debian10 64bit

slabua, 2021-08-14:
If all the rest is correctly configured, you might just have to set /etc/resolv.conf to something valid.

xSetBack, 2021-08-19:
> If all the rest is correctly configured, you might just have to set /etc/resolv.conf to something valid.

I've solved using this script https://linux-pwnagotchi-conn-share.sourceforge.io/ that seems the original bash script for connection sharing with some modification made by https://t.me/xNigredo

closed 2021-08-19 by xSetBack