evilsocket/pwnagotchi #1008: how did you make the bootable pwnagotchi image?

bwanaaa, 2021-06-14:
The latest code in the github is just the code. How did you make the live image that gets flashed to the sd card?

grokbeer, 2021-06-15:
That's all done under the builder subdirectory. Look at the top-level Makefile under the image target to see how it's built.

wavesailor, 2022-07-28:
Does anyone have detailed instructions on how to build an image from source?

wavesailor, 2022-09-20:
I have tried numerous times - on a fresh PI OS Lite and on Ubuntu 22.04 but I get a ton of errors. 
Does anyone have a script? or maybe a docker container script that I can use to build and image?

Ra2-IFV, 2022-09-24:
> I have tried numerous times - on a fresh PI OS Lite and on Ubuntu 22.04 but I get a ton of errors. Does anyone have a script? or maybe a docker container script that I can use to build and image?

Show me the errors. I think I can fix some of them.
Or you can check my fork and try to build your image with it.

Sirius-RJ, 2022-11-06:
I guess i just dont understand how to launch it, is bettercap initially the launchpad, trying for running on kali no pi, seems impossible but people say it is. im kinda new to all pwnagotchis and just wanted to build one with out the hardware to see if it was worth building a portable one.