evilsocket/pwnagotchi #1012: help understanding the wpa handshakes that pwnagotchi captures

bwanaaa, 2021-06-18:
I tried to test some  handshakes that i captured but had no luck with hashcat. So I thought I would try them using https://www.onlinehashcrack.com/wpa

I submitted the pcap file but that site said:

"No valid EAPOL handshake or PMKID found in the submitted file."

I am new to testing this stuff. I captured some handshakes with a pwnagotchi and others using wifite in kali.

How can I tell if the pcap file (or the converted hccapx file) is worth using against hashcat? If pwnnagotchi captures something, does that mean it is a valid handshake if I find it in the handshakes directory of the pwnagotchi?

stylesuxx, 2022-01-03:
> "No valid EAPOL handshake or PMKID found in the submitted file."

From what I have seen, this might happen if you capture a partial handshake, device tried to authenticate with a wrong password for example.

Generally speaking when you convert it and a valid handshake is found, it is "worth" trying to crack it.

undermuz, 2022-02-15:
i also faced with this problem
but if `set wifi.handshakes.aggregate false; set wifi.handshakes.file ~/bettercap-wifi-handshakes.pcap` in bettercap, and send bettercap-wifi-handshakes.pcap to onlinehashcrack, it submitted success
seems like bettercap bug