evilsocket/pwnagotchi #1015: [BUG] pwnagotchi numpy error

Luca-Hiller, 2021-06-22:
### Hello, I read in a post that you can turn on the AI by uninstalling numpy and reinstalling it, this error occurred after the reinstallationI don't know how to fix this but maybe someone can help me.

**(See image)**

_Greetings from Germany_

[image]

[image]

 - Pwnagotchi version 1.5.5
 - Type of hardware Pi W

iproxy, 2021-06-28:
What version did you have before 1.5.5? I was digging around and saw:
Installation of 1.5.3 and doing an update is working. 

You can also try:
pip uninstall numpy
pip install numpy==1.17.2

If Python3 > try **pip3**

pip3 uninstall numpy
pip3 install numpy==1.17.2

Since I haven't updated mine for months, I'll give it a shot and report back.

Luca-Hiller, 2021-06-29:
That does not work with pip3 install numpy==1.17.2 but i will try to downgrade it to 1.5.3 and Upgrade it back to 1.5.5

but thanks anyway

iproxy, 2021-06-29:
> That does not work with pip3 install numpy==1.17.2 but i will try to downgrade it to 1.5.3 and Upgrade it back to 1.5.5
> 
> but thanks anyway

I have to actually upgrade to the latest myself. I haven't upgraded for quite some time (months) I'm a bit far behind. I'll do it later today and see if I run into any issues. If I do, I'll mess around with it until I get it working then provide a step-by-step solution.

peterus, 2021-11-12:
@evilsocket maybe it is time for a new release where this issue is finally fixed. It is really hard for newcomers (me included) to fix this by hand.
also this issue https://github.com/evilsocket/pwnagotchi/issues/1046 would be fixed with a new release.

evilsocket, 2021-11-12:
@peterus did somebody debug and fix this and is a PR available for merge? i'm just [one of 133 contributors](https://github.com/evilsocket/pwnagotchi/graphs/contributors), i'll gladly merge a fix when available but being this an unpaid project I do not have the tiime to work on it myself right now.

peterus, 2021-11-12:
@evilsocket totaly understand! I have also a lot of projects and I am not getting paid 😂
I was reading about this issue now in several tickets and the only solution is to use a newer numpy version (deinstall the current and install the latest numpy).

I will test on my development machine to update numpy, test it on my raspberry zero and create a MR when done.

skontrolle, 2021-11-13:
> I have to actually upgrade to the latest myself. I haven't upgraded for quite some time (months) I'm a bit far behind. I'll do it later today and see if I run into any issues. If I do, I'll mess [...]

I had to update to latest to get it to work too.  1.17.2 doesn't work after the `apt full-upgrade`.
 `sudo pip3 install --upgrade numpy`

debian-user-france1, 2021-12-05:
Still not got fixed.. Project is idle - 8 Months since last update in the master branch

donaciano, 2022-06-28:
> I had to update to latest to get it to work too. 1.17.2 doesn't work after the `apt full-upgrade`. `sudo pip3 install --upgrade numpy`

This fixed it for me. After updating everything I could with apt and doing all the suggested things near the top what finally got the AI to load for the first time on my Pi Zero W was:
```
/etc# pip3 install --upgrade numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy
pwnagotchi 1.5.5 has requirement numpy==1.17.2, but you'll have numpy 1.21.6 which is incompatible.
matplotlib 3.4.1 has requirement pillow>=6.2.0, but you'll have pillow 5.4.1 which is incompatible.
Installing collected packages: numpy
  Found existing installation: numpy 1.17.2
    Uninstalling numpy-1.17.2:
      Successfully uninstalled numpy-1.17.2
Successfully installed numpy-1.21.6
```

Now you'd think the warning means the numpy would break however using tail -f on my /var/log/pwnagotchi.log I finally saw these:
```
[INFO] [ai] bootstrapping dependencies ...
<snip 7 minutes>
[INFO] [ai] creating model ...
<30-35 min after boot>
[ai] model created:
[ai] learning for 50 epochs ...
```
After waiting a while to make sure the incompatible numpy version didn't actually break things I started getting handshakes.
Thanks for the fix!  Hope it works for others.

Nannk, 2023-07-15:
This problem can indeed be solved by updating numpy. Ive ran `sudo pip3 install --upgrade numpy` without updating pip. It installed numpy-1.21.6. The numpy build process took about 1.5 hours on RP0W. The ai seems to be working fine.
[image]

buzibu, 2023-12-06:
> On 1.5.5 this problem can indeed be solved by updating numpy. Ive ran `sudo pip3 install --upgrade numpy` without updating pip. It installed numpy-1.21.6. The numpy build process took about 1.5 hours on RP0W. The ai seems to be working fine.
> 
> Edit: sudo is quite important in this case ![image](https://private-user-images.githubusercontent.com/34575742/253753789-e98ff750-627b-475b-a667-4b6b97df5077.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4OTE0MTEsIm5iZiI6MTcwMTg5MTExMSwicGF0aCI6Ii8zNDU3NTc0Mi8yNTM3NTM3ODktZTk4ZmY3NTAtNjI3Yi00NzViLWE2NjctNGI2Yjk3ZGY1MDc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA2VDE5MzE1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2ZGNmNzhkMWRiMTA5NDUxNGVhODMxZDk4ODhjMzkzOWJmNWVhMWE1ODRhNmU5NzlmODRjNDJjOWUyNGM5NWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZbdfT-Gsk6NnD4aKN40HvDY0EorzS5LX6etbR4fkw-g)

I also installed numpy-1.21.6 the same way, but the error did not disappear. Did you do something else before updating numpy?

TobiStacks, 2023-12-27:
I am also getting this issue on v1.5.5 of pwnagotchi. I attempted installations using both root and sudo, using pip3 install --upgrade numpy, but still getting the same issue

Luca-Hiller, 2024-01-01:
> I am also getting this issue on v1.5.5 of pwnagotchi. I attempted installations using both root and sudo, using pip3 install --upgrade numpy, but still getting the same issue

Its a long time ago where i had this problem i fixed it but dont know how i did it maybe try asking in the Unofficial Discord Server

https://discord.gg/SWUvXN4e