evilsocket/pwnagotchi #1016: [BUG]

TTsdzb, 2021-06-28:
**Describe the bug**
Pwnagotchi stucks in a loop of exception, after getting a handshake.

**To Reproduce**
Steps to reproduce the behavior:
1. Wait for pwnagotchi get a handshake.
2. Randomly, after the capture its screen doesn't change any more.
3. Now login via ssh and view the log, it is printing an exception again and again, and doesn't work anymore

**Logs**
[2021-06-28 23:22:47,483] [INFO] error 400: wifi.interface not set or not found
[2021-06-28 23:22:47,487] [ERROR] main loop exception (error 400: wifi.interface not set or not found)
Traceback (most recent call last):                                  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/bettercap.py", line 11, in decode
    return r.json()
File "/usr/lib/python3/dist-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)                               File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())                simplejson.errors.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 55, in do_auto_mode
    agent.recon()
File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/agent.py", line 157, in recon
    self.run('wifi.recon.channel clear')
File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/bettercap.py", line 55, in run
    return decode(r, verbose_errors=verbose_errors)                 File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/bettercap.py", line 19, in decode
    raise Exception(err)
Exception: error 400: wifi.interface not set or not found

**Environment**
 - Pwnagotchi version: 1.5.5
 - OS version
 - Type of hardware: Pi0W
 - Any additional hardware use: An waveshare epd, no additional network interfaces

Luca-Hiller, 2021-06-29:
Maybe try reinstalling your Pwnagotchi on your SD card as it looks like a WiFi interface problem that I never had, if that doesn't work try another version of Pwnagotchi like 1.5.3

TTsdzb, 2021-06-29:
In fact I flashed 1.5.3 and it automatically updated itself to 1.5.5

jonschnell, 2022-06-01:
I am seeing the same log entry and behavior intermittently on 1.5.5 Pi0W.