evilsocket/pwnagotchi #1018: Unable to ssh and perform basic pwnagotchi functions

Decapitater, 2021-07-10:
Hi,
I'm unable to ssh and perform basic functionalities such as bettercap login 
i created the config.toml file with Bluetooth+initial configs however after the first boot there were no changes to location of config file and there is not folder named pwnagotchi in /etc/pwnagotchi/config.toml and so file is also missing there 
my machines recognises it as ethernet and i have changed the ip , netmask , gateway to required ones 
i can ping both 10.0.0.1 and 10.0.0.2 but the ssh is not working for 0.2 and showing wrong password for 0.1 
i have tried 2 different linux systems and 3 different cables all with power+data 
once when i tried i was able to enter bettercap but through 10.0.0.2 in browser and not using pwnagotchi.local 
I tried the host sharing script and the script always indicates the file exist

gili-gili, 2021-07-19:
Hummmm,i think you can reinstall the system.It might work.

Decapitater, 2021-07-22:
i tried 5 times with 2 different os no success i tried to use rpi imager too

gili-gili, 2021-07-23:
What does your computer recognize your pi?If there's nothing wrong with the driver,it should be recognize like this
[image]
And then set up it.
[image]
Then you should connect to your pi
[image]

Decapitater, 2021-07-23:
sorry forgot to tell im on linux
its recognised as usb ethernet after 10.0.0.1 and mask procedure 
however i cannot access ssh i can ping both domains 10.0.0.1 and 10.0.0.2
the bettercap sits on 10.0.0.1

gili-gili, 2021-07-23:
So you may have installed the wrong driver automatically.Unfortunately i only have Windows driver:(
Why not try Windows?
Another possibility is that you have written the wrong config.toml.

Decapitater, 2021-07-23:
hmm any ideas on config.toml i tried many different ways to tackle it 
everytime the results were unique and sometimes they were straightup non functional 
i will try it with windows 
so would it work on my linux machine afterwards i wonder
thanks

gili-gili, 2021-07-23:
You can follow
[https://pwnagotchi.ai/configuration/](url)
to configure it
And you probably need this driver
[https://drive.google.com/file/d/1UNy60lKIh7it-3gqsMJXutUkSxwebzKj/view?usp=sharing](url)
Or your pi W will be recognized like this
[image]

Decapitater, 2021-07-24:
@evilsocket  somehelp needed

Sundancer989, 2021-07-27:
Have you tried adding a blank ssh txt file to the boot partition to enable
ssh on first boot? Can you run nmap on the two IPs to confirm if the port
is open and still 22 or whatever it may have been changed to?

Decapitater, 2021-07-29:
to confirm there is a ssh file that is blank. I haven't checked it with nmap.

Decapitater, 2021-08-01:
Interesting finding 
I'm was able to ssh and make a working pwnagotchi with help of windows.
It seems to me that the linux working is not so great as i was unable to ssh on a linux comp even after making everything sure.
The build is now working.
Thanks @gili-gili  and @Sundancer989  for support

closed 2021-08-01 by Decapitater

WudUpA, 2024-04-15:
What did you use for the default password after 
ssh pi@10.0.0.2
The default password should be raspberry....right?