evilsocket/pwnagotchi #1020: [BUG] Not connected to Windows

jannikb16, 2021-07-15:
**Describe the bug**
The Pwnagotchi is not displayed in the device manager

**Expected behavior**
I reinstalled the Pwnagotchi and wanted to connect it to my computer. The Pwnagotchi is not displayed in the device manager

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.5.5
 - OS version: Windows 10
 - Type of hardware: RasPi Zero W Rev. 1.1

gili-gili, 2021-07-19:
You need a driver.
[https://drive.google.com/file/d/1UNy60lKIh7it-3gqsMJXutUkSxwebzKj/view?usp=sharing](url)
Select COM1 and install it.

jannikb16, 2021-07-19:
COM1 is an ESP. It's not the raspberry

gili-gili, 2021-07-23:
Oh i see.....
Please check your ports.There should be 2 ports on your pi0W,one is PWR IN and another is USB.Make sure you connect your pi from the USB port.
And Cheaper quality micro-USB cords often do not support data transfer and will NOT allow you to actually connect to your Pwnagotchi.