evilsocket/pwnagotchi #1023: web ui not loading with bluetooth

LuxStatera, 2021-07-21:
raspberry pi zero w
pwnagotchi 1.5.5
internet sharing between pwnagotchi and computer works but doesnt seem to work with bt-tethering. it shows that they are paired and connected but the web ui will not load

BarnOwlDown, 2021-07-22:
I have this problem but only on Raspberry Pi 3B

wpa-3, 2021-07-27:
It works fine for me, you sure your settings are correct? 
What phones are you using?

JettScythe, 2021-09-11:
I'm also having this issue. Fresh install of 1.5.3 and upgrade to 1.5.5. I can still see the web-ui when plugged into the data port on a computer, but the bt-tethering web-ui seemed to only work a single time.

makerph, 2021-09-11:
be sure to have the port # at the end :8080

JettScythe, 2021-09-11:
I had the same issue with a fresh 1.5.3 install that auto-updated to 1.5.5
Weirdly enough, my issue appears to be the power bank I was using. If its not at > 70% charge the web ui is unstable or just doesnt load.

TaylorHassall, 2021-09-23:
Also had the same issue after updating from 1.5.3 with an auto update - reformatted to 1.5.5 and it worked without issue.