evilsocket/pwnagotchi #1024: put pwnagotchi in french

coutud, 2021-07-27:
hi
i don't know how and witch code i must write to have it in french

thanks for your help

Decapitater, 2021-08-02:
ami cela devrait aider , Modifier la langue dans config.toml voir les captures d'écran
[image]
[image]

coutud, 2021-08-02:
bonjour

merci de ta réponse mais j'ai vu cela sauf que je ne sais pas les commandes pour le faire car lorsque dans le fichier config je mets     fr au lieu de en mais cela ne marche pas

Decapitater, 2021-08-02:
qu'est-ce qu'il montre ?

coutud, 2021-08-02:
Il parle anglais et non français

Decapitater, 2021-08-02:
n'importe quel journal de langue?

coutud, 2021-08-02:
comment ca ?

dis moi comment tu fais pour mettre ne Français et je ressaye ce soir et te dis

merci

coutud, 2021-08-05:
bonjour,

si je mets le fichier config comme suit est ce ok

main.name = "cougar"
main.lang = "fr"

main.custom_plugins = ""
main.custom_plugin_repos = [
 "https://github.com/evilsocket/pwnagotchi-plugins-contrib/archive/master.zip"
]

main.whitelist = [
 ""
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
 ""
]

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.net-pos.enabled = false
main.plugins.net-pos.api_key = "test"

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.webgpsmap.enabled = false

main.plugins.onlinehashcrack.enabled = false
main.plugins.onlinehashcrack.email = ""
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false
main.plugins.onlinehashcrack.whitelist = []

main.plugins.wpa-sec.enabled = true
main.plugins.wpa-sec.api_key = ""
main.plugins.wpa-sec.api_url = "https://wpa-sec.stanev.org"
main.plugins.wpa-sec.download_results = true
main.plugins.wpa-sec.whitelist = [""]

main.plugins.wigle.enabled = false
main.plugins.wigle.api_key = ""
main.plugins.wigle.whitelist = [""]

main.plugins.bt-tether.enabled = true

main.plugins.bt-tether.devices.android-phone.enabled = false
main.plugins.bt-tether.devices.android-phone.search_order = 2
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.0.3"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
main.plugins.bt-tether.devices.android-phone.share_internet = false
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = true
main.plugins.bt-tether.devices.ios-phone.search_order = 1
main.plugins.bt-tether.devices.ios-phone.mac = "'20:E2:A8:24:BE:74'"
main.plugins.bt-tether.devices.ios-phone.ip = "192.168.0.3"
main.plugins.bt-tether.devices.ios-phone.netmask = 14
main.plugins.bt-tether.devices.ios-phone.interval = 1
main.plugins.bt-tether.devices.ios-phone.scantime = 15
main.plugins.bt-tether.devices.ios-phone.max_tries = 0
main.plugins.bt-tether.devices.ios-phone.share_internet = true
main.plugins.bt-tether.devices.ios-phone.priority = 999

main.plugins.memtemp.enabled = true
main.plugins.memtemp.scale = "celsius"
main.plugins.memtemp.orientation = "horizontal"

main.plugins.paw-gps.enabled = true
main.plugins.paw-gps.ip = ""

main.plugins.gpio_buttons.enabled = false

personality.advertise = true
personality.deauth = false

ui.fps = 0.0
ui.font.name = "DejaVuSansMono" # for japanese: fonts-japanese-gothic
ui.font.size_offset = 0 # will be added to the font size

ui.web.enabled = true
ui.web.address = "0.0.0.0"
ui.web.username = ""
ui.web.password = ""
ui.web.origin = ""
ui.web.port = 8080
ui.web.on_frame = ""

ui.display.enabled = true
ui.display.rotation = 2
ui.display.type = "waveshare_2.3"
ui.display.color = "black"

merci
----- Mail original -----

Decapitater, 2021-08-06:
que diriez-vous d'essayer le config.toml minimum nu
main.name = "Cougar"
main.lang = "fr"
main.whitelist = [
  "EXAMPLE_NETWORK",
  "ANOTHER_EXAMPLE_NETWORK",
  "fo:od:ba:be:fo:od",
  "fo:od:ba"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "YourHomeNetworkHere"
]

ui.display.enabled = true
ui.display.type = "waveshare_2"
ui.display.color = "black"

la recherche de l'appareil fonctionne-t-elle correctement?
est config.toml placé dans la partition de démarrage après avoir fait la carte SD ?
Quelque chose de bizarre qui n'était pas censé se produire?

coutud, 2021-08-06:
Bonjour 
J’essaie aujourd’hui et te dis
Que veux tu dire par recherche de l’appareil

Je voulais me connecter en Bluetooth dessus c’est pourquoi je voulais mettre le fichier avec le parametrage sur iOS par contre l’adresse ip est elle a modifiée

Le fichier config est bien sur la carte sd 

Merci

Decapitater, 2021-08-06:
profitez de votre temps avec pwnagotchi alors terminez le problème?

coutud, 2021-08-06:
Bonjour
L’image et la config marche en français

Merci

Decapitater, 2021-08-07:
Bonjour 
profiter de votre pwnagotchi f
ermer le problème?

coutud, 2021-08-07:
Bonjour,

 

Oui on peut fermer le sujet

 

Merci

Decapitater, 2021-08-07:
derek close

Decapitater, 2021-08-07:
fermez le problème maintenant en utilisant le bouton?

closed 2021-08-07 by coutud