evilsocket/pwnagotchi #1025: [BUG] .pwnagotchi-auto gets deleted automatically

jesseuwu, 2021-08-03:
**Describe the bug**
I have my pwnagotchi plugged in via the data USB port. I want it to automatically start in AI mode, so I created a `/root/.pwnagotchi-auto` file. The next reboot, it will start in AI, but then the file gets deleted and the following reboot will be in Manual mode again.

**To Reproduce**
Steps to reproduce the behavior:
1. run `sudo touch /root/.pwnagotchi-auto`
2. reboot system. system reboots into AI mode.
3. reboot again. file is deleted and system goes to Manual mode.

**Expected behavior**
The file should stay there and pwnagotchi starts in Auto/AI mode until the file is manually deleted.

**Environment (please complete the following information):**
 - v.1.5.5
 - RPi0

rehatiel, 2021-08-07:
Confirmed this happens to me too.

jesseuwu, 2021-08-08:
https://github.com/evilsocket/pwnagotchi/blob/0704541dd19401229138d0e8a8f313a5904e2de9/builder/data/usr/bin/pwnagotchi-launcher#L15

I think changing this function call from `is_auto_mode` to `is_auto_mode_no_delete` would fix it. They are the same function, just one deletes the file and one doesn't.

===

Was `.pwnagotchi-auto` intended to be a single-use file that gets deleted every restart? The page at https://pwnagotchi.ai/usage/ makes it seem like it's supposed to persist. In case it _is_ intended as a single-use file, what's the correct way to force my device to always start in Auto mode despite using the data USB port? 

> You can "force" the unit to **always** go in AUTO mode regardless of which USB port you're using by creating the /root/.pwnagotchi-auto file.

TQMatvey, 2022-03-09:
bump, happens to me too

fwartner, 2023-12-30:
Still present.

HanSyt, 2025-05-29:
Still present..
chattr +i .pwnagotchi-auto could be  nice solution