evilsocket/pwnagotchi #1028: [BUG] UPS-Lite not working properly

hexan86, 2021-08-08:
Pwnagotchi is showing only a number followed by a dash.
i.e.: 63-

My steps:
1. Shorted the two pads
2. Enabled i2c
3. Followed instructions for i2cdetect
4. Tried also to use the ups_lite.py from linshuqin329

I was expecting the standard Voltage & Percentage

https://ibb.co/vsFndXV

Environment:
 - Pwnagotchi v1.5.5
 - Raspberry Pi Zero W 
 - UPS-Lite v1.2
 - PCF8523
 - Waveshare v2

**Additional context**
With the original plugin I was expecting the standard V/%, instead I ended up with an "xx-". I tried to substitute /usr/local/src/pwnagotchi/pwnagotchi/plugins/default/ups_lite.py with another one, but I still have "xx-".
I am unable to find what is putting at screen this result.

Xbeed, 2021-08-13:
I'm having the same problem. [My reddit post](https://www.reddit.com/r/pwnagotchi/comments/o6y9ho/ups_lite_12_showing_plus_instead_of_percentage_i/)
The UPS-Lite plugin in Pwnagotchi is for v1.1 if i not mistaken.

hexan86, 2021-08-13:
Yours is indicating the battery value like increasing, instead of
discharging?

Xbeed, 2021-08-14:
it increases to how much the battery has been charged. After that, it will display "xx-" while the battery is discharge.

hexan86, 2021-08-14:
Have you shorted the pads behind?
I did and now it goes down (66-, 65-, 64-,...), But still with some
problems...like I've fully charged it, but starts from 73 and not from 100.

Xbeed, 2021-08-15:
Yes.  I shorted the 2 pads behind. 
I'm having the same problem! No matter how long I charge, it will start from 73. And it will decrease with "xx-"

najashark, 2022-08-29:
I also have the same issue, can you do ```sudo i2cdump -y 1 0x32``` and ```sudo i2cdump -y 1 0x36``` for me

hexan86, 2022-08-29:
> I also have the same issue, can you do `sudo i2cdump -y 1 0x32` and `sudo i2cdump -y 1 0x36`

https://pastebin.pl/view/bf075f72

najashark, 2022-08-29:
from your dump, it should be good. it shows all required data for ups-lite.py to function. did you still having issue?

hexan86, 2022-08-30:
> from your dump, it should be good. it shows all required data for ups-lite.py to function. did you still having issue? because mine look like this, I suspect that I got a defected product
> 
> ```
> pi@pwnagotchi:~ $ sudo i2cdump -y 1 0x32
> No size specified (using byte-data access)
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
> 00: ff ff 00 00 00 00 00 00 00 02 ff ff 97 00 ff ff    .........?..?...
> 10: ff ff ff ff 00 00 ff ff ff ff ff ff ff ff ff ff    ................
> 20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
> 30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 00    ................
> ```

I've noticed that mine works better if I proceed in this way:
Power on the ups-lite, connect the charging cable, wait 'til it reaches 100%, turn off the ups-lite. 
Now, if I turn it on again, it goes his way from 100 to 1 without issues,  but if I turn it off midway and put it in charge, it scrambles again the charge level.