evilsocket/pwnagotchi #1031: Radxa Zero

flokiauther, 2021-08-18:
What about Radxa Zero support?

lolzdatguy, 2021-08-18:
You could try building yourself

aazarmemon, 2021-08-28:
Will look into possible port once I get my hands on one, Supposed to be mailed out this weekend but I smell delays.

makerph, 2021-09-03:
> Will look into possible port once I get my hands on one, Supposed to be mailed out this weekend but I smell delays.

what version did you choose? Let us know if all work out. thanks

aazarmemon, 2021-09-03:
> > Will look into possible port once I get my hands on one, Supposed to be mailed out this weekend but I smell delays.
> 
> what version did you choose? Let us know if all work out. thanks

4GB LPDDR4 RAM, WIFI5/BT5, 128GB eMMC

 
Just Shipped Today.

aazarmemon, 2021-09-21:
Still waiting for the mail...  Got the 128GB 3MMC version.  If you have one already, you could try the install method outlined on the website (https://pwnagotchi.ai/installation/#installing-on-any-gnu-linux) using Radxa Zero SBC compatible Linux.  Failing that, you might have to do a complete rebuild.

trwbox, 2022-06-19:
I was wondering if there were any updates on if you got it working?

5p0ck, 2022-09-26:
Not yet, ran out of time, got back to work

micklen, 2025-04-24:
Any progress with Radxa Zero?