evilsocket/pwnagotchi #1038: Not possible to enter AI mode 1.5.5

gui28347, 2021-09-17:
Not possible to enter AI mode in 1.5.5

## Expected Behaviour
Enter in AI mode after 10~15 minutes

## Current Behaviour
Failing to enter AI mode

## Possible Solution

## Steps to Reproduce (for bugs)
Easy to repro:
1. Install 1.5.5
2.  apt-get update & apt-get upgrade
3.  Restart with cable to enter in AI mode
4.  Wait 6 minutes, and check the logfiles.

## Context
Not possible to fix.
Logfiles show:

[2021-09-17 19:24:23,009] [ERROR] error while starting AI (libavcodec.so.58: cannot open shared object file: No such file or dire                                                    ctory)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/__init__.py", line 21, in load
    from stable_baselines import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/__init__.py", line 7, in <module>
    from stable_baselines.gail import GAIL
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/gail/__init__.py", line 2, in <module>
    from stable_baselines.gail.dataset.dataset import ExpertDataset, DataLoader
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/gail/dataset/dataset.py", line 5, in <module>
    import cv2
  File "/usr/lib/python3/dist-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: libavcodec.so.58: cannot open shared object file: No such file or directory

## Your Environment

- [ Yes 1.5.5 ] You're using the official images

- [ Yes ] You're using a raspberry pi 0
  
- [Yes waveshare_2 ] You're using a supported LCD

From previous issues I found :

sudo apt update && apt upgrade -y
sudo pip3 install --upgrade pwnagotchi

And Also 

pip3 uninstall numpy
pip3 install numpy
 sudo apt install libatlas3-base -y

Tried and same error still happen.

Vector45555, 2021-09-22:
I believe this may solve the problem

[https://stackoverflow.com/questions/66060487/valueerror-numpy-ndarray-size-changed-may-indicate-binary-incompatibility-exp](https://stackoverflow.com/questions/66060487/valueerror-numpy-ndarray-size-changed-may-indicate-binary-incompatibility-exp)

Vector45555, 2021-09-22:
> I believe this may solve the problem
> 
> https://stackoverflow.com/questions/66060487/valueerror-numpy-ndarray-size-changed-may-indicate-binary-incompatibility-exp

Although I still haven't connected mine to internet properly, so mine is still not working nor can I test if if works

gui28347, 2021-09-22:
I was able to execute
pip3 uninstall numpy
pip3 install numpy

And it fixed the issue. The caveat is that I didn't use the apt-get update & apt-get upgrade.

So appears that as long as you don't update all other packages, only upgrading this package its possible workaround.

closed 2021-09-22 by gui28347

Katazui, 2022-03-24:
I know this closed, but I ran into the same error while trying to get 1.5.5 to work. I am happy to say that AI now works after running

sudo pip3 uninstall numpy
sudo pip3 install numpy
sudo apt install libavformat-dev (<- this code was the one that help me fix the error: [Reddit Post Fix](https://www.reddit.com/r/pwnagotchi/comments/pt2fs1/comment/hgmcc3x/?utm_source=share&utm_medium=web2x&context=3))

Katazui, 2022-05-18:
@aidenmitchell Mine took about 2hrs on RPi0.

crashlogs, 2023-12-26:
> I know this closed, but I ran into the same error while trying to get 1.5.5 to work. Hopefully, I can help someone. I am happy to announce that AI now works after running...
> 
> sudo pip3 uninstall numpy sudo pip3 install numpy sudo apt install libavformat-dev (<- this code was the one that help me fix the error: [Reddit Post Fix](https://www.reddit.com/r/pwnagotchi/comments/pt2fs1/comment/hgmcc3x/?utm_source=share&utm_medium=web2x&context=3))

Life saving, I've powered up my 3 year old pwnagotchi today and it was stuck, neural network not starting, all I did was this and it worked perfectly

Still hesistating about apt update & upgrade, since it working I might keep it how it is