evilsocket/pwnagotchi #1039: waveshare_3 support?

NA3, 2021-09-18:
i just notice my wavashare is a v3
the ink display dont initialise, someone can help please?

stylesuxx, 2022-01-03:
Try using this config:

```
ui.display.enabled = true
ui.display.type = "waveshare213d"
```

elade, 2022-01-05:
> Try using this config:
> 
> ```
> ui.display.enabled = true
> ui.display.type = "waveshare213d"
> ```

Not working for me with V3.
I've also tried to run the waveshare examples from their git (https://github.com/waveshare/e-Paper) and it worked only with  epd_2in13_V3_test.py

ikornaselur, 2022-01-10:
Another V3 that doesn't work with `waveshare213d` as the type unfortunately. Annoyingly the listing where I bought the screen specifically says `V2` and even has the photos showing `V2` sticket, but the arrived displays are V3 Rev 2.1.

tdvantine, 2022-06-11:
@ikornaselur maybe fork the repo with these fixes since it seems this repo isn't updated anymore haveing close to 200 open issues and this not being merged for almost a year.

lcrince, 2022-06-20:
Hey I also have a V3, but I'm a noob at this and I'm not sure how to proceed. I found this merge, but i'm not sure how to implement it.
can anybody help?

closed 2022-07-18 by evilsocket

mifkilla, 2023-06-22:
https://pwnsec.io/how-to-make-a-pwnagotchi-in-2022-with-a-waveshare-v3-screen/