evilsocket/pwnagotchi #1045: [BUG] error while starting AI (numpy.ndarray size changed, may indicate binary incompatibility.

michelep, 2021-11-01:
**Describe the bug**
In pwnagotchi.log i found this error: 

```
[ERROR] error while starting AI (numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/__init__.py", line 21, in load
    from stable_baselines import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/__init__.py", line 1, in <module>
    from stable_baselines.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/__init__.py", line 1, in <module>
    from stable_baselines.a2c.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 6, in <module>
    import tensorflow as tf
  File "/usr/lib/python3/dist-packages/tensorflow/__init__.py", line 98, in <module>
    from tensorflow_core import *
  File "/usr/lib/python3/dist-packages/tensorflow_core/__init__.py", line 35, in <module>
    from tensorflow._api.v1 import compat
  File "/usr/lib/python3/dist-packages/tensorflow_core/_api/v1/compat/__init__.py", line 21, in <module>
    from tensorflow._api.v1.compat import v1
  File "/usr/lib/python3/dist-packages/tensorflow_core/_api/v1/compat/v1/__init__.py", line 652, in <module>
    from tensorflow_estimator.python.estimator.api._v1 import estimator
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/__init__.py", line 8, in <module>
    from tensorflow_estimator._api.v1 import estimator
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/_api/v1/estimator/__init__.py", line 8, in <module>
    from tensorflow_estimator._api.v1.estimator import experimental
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/_api/v1/estimator/experimental/__init__.py", line 8, in <module>
    from tensorflow_estimator.python.estimator.canned.dnn import dnn_logit_fn_builder
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/__init__.py", line 25, in <module>
    import tensorflow_estimator.python.estimator.estimator_lib
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/estimator_lib.py", line 53, in <module>
    from tensorflow_estimator.python.estimator.inputs import inputs
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/inputs.py", line 22, in <module>
    from tensorflow_estimator.python.estimator.inputs.numpy_io import numpy_input_fn
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/numpy_io.py", line 26, in <module>
    from tensorflow_estimator.python.estimator.inputs.queues import feeding_functions
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/queues/feeding_functions.py", line 40, in <module>
    import pandas as pd
  File "/usr/local/lib/python3.7/dist-packages/pandas/__init__.py", line 29, in <module>
    from pandas._libs import hashtable as _hashtable, lib as _lib, tslib as _tslib
  File "/usr/local/lib/python3.7/dist-packages/pandas/_libs/__init__.py", line 13, in <module>
    from pandas._libs.interval import Interval
  File "pandas/_libs/interval.pyx", line 1, in init pandas._libs.interval
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject

```

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.5.5
 - Type of hardware: Rpi W Zero

nfcariello, 2022-01-05:
@michelep I was able to resolve this error by running
`sudo pip3 uninstall numpy`
`sudo pip3 install numpy==1.20.2`

for some reason it’ll crash the whole thing and won’t start if you use the latest numpy, but 1.20.2 seems to do the trick, I hope this helps!

HighMarck17, 2022-04-24:
Also `sudo pip3 install -U numpy` worked for me.