evilsocket/pwnagotchi #1046: Image won’t boot on Raspberry Pi Zero 2 W

AlexWhitehouse, 2021-11-02:
## Expected Behaviour
Expected to boot on Pi Zero 2 W. Fails to boot.

## Current Behaviour
Fails to boot

## Possible Solution
Recompile kernel for RPiZ2W

## Steps to Reproduce (for bugs)
1. Flash image to SD
2. Load SD into RPiZ2W
3. Try to boot

## Your Environment

- [ ] You're using the official images (1.5.5)

- [ ] You're using a raspberry pi zero 2 w

- [ ] You're using a supported LCD (waveshare v2)

akhepcat, 2021-11-02:
Funny - i just swapped mine over with no issues at all:

root@pwnagotchu:~# dmesg|more
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.83-Re4son-v7+ (root@kali) (gcc version 10.2.1 20210110 (Debian 10.2.1-6)) #1 SMP Sat May 15 03:22:58 UTC 2021
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 Rev 1.0

root@pwnagotchu:~# lscpu
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1

root@pwnagotchu:~# cat  /usr/local/src/pwnagotchi/pwnagotchi/_version.py 
__version__ = '1.5.5'

AlexWhitehouse, 2021-11-03:
I don't even get past the multicoloured square to boot to the kernel image. Neither do I get any status lights on the pi itself. I have loaded other images to check there is no problem with the pi and these have booted successfully.

akhepcat, 2021-11-03:
FWIW - i'm running on a 64G  SanDisk Ultra,  Class 10 (A1)    that had previously been running on a Pi0w-v1

You could try booting up on a  V1,  using the usb ethernet sharing and "internet sharing"   then ssh into the pi and make sure there are no filesystem issues, and run an apt-get update / upgrade  cycle to get everything up to date?  (you may need to manually fix-up /etc/resolv.conf,  and set your time/date.)

But I didn't do any of that prior to booting it on the V2 ... just swapped the card over, and it worked fine.

AlexWhitehouse, 2021-11-03:
Very strange, I don't have a v1 to hand to test that unfortunately. Does the fact there are no status lights or anything output to HDMI suggest that the pi is not accepting the image on the SD card? It works fine with other builds just not the pwnagotchi image.

akhepcat, 2021-11-03:
you could try re-imaging the sdcard.    Or try booting it on any other pi hardware you might have, possibly excepting for the wifi and cdcether, it should otherwise start up  (tested on a Pi4, even cdcether  works)

AlexWhitehouse, 2021-11-03:
I have re-imaged the sdcard, plugged into RPi Zero 2 and nothing. I then plug the same card into a Pi4 and I get activity lights and an IP address. I then plug the same card back into the zero 2 and get nothing again. I have also tried just raspbian light on the SD card in the zero 2 and this works fine.

Junohea, 2021-11-05:
Seeing the same issue. The Zero 2 fails to boot with any of the following:
- clean flash of v1.5.5
- clean flash of v1.5.5, apt update&upgrade 
- "old image" (running latest 1.5.5)
- old image, apt update&upgrade 

But it did boot with the latest raspbian

For reference, here's what we see while it boots from the Pi Zero W (non Zero2)
root@pwnagotchi:~# dmesg|more
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.83-Re4son+ (root@kali) (gcc version 10.2.1 20210110 (Debian 10.2.1-6)) #1 Sat May 15 12:50:45 UTC 2021
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1

isthisausername1, 2021-11-05:
I am having the same problem on my zero 2.
I have also tried the steps @Junohea tried in his post above.
The sd card works fine on my rpi 4 and everything else with it works to but the status lights don't even blink on the zero2.

akhepcat, 2021-11-05:
One difference i see in your bootlog  is  the kernel version:
Mine (booting) : 5.4.83-Re4son-v7+
Yours (failing) : 5.4.83-Re4son+

This would seem to indicate that you need to upgrade your kernel, or the image that  you're using.

Boot it on a working device,  get internet connectivity (via wired, or 2nd wifi)   and run a distro upgrade to see if there's an upgraded kernel for you. 

(apparently I may have done that, but I don't *remember* upgrading the system...)

isthisausername1, 2021-11-05:
I think it defiantly could have been that, unfortunately i didn't check the version before updating but now after doing

sudo apt update && sudo apt full-upgrade -y

it started working and now it boots and i can ssh into it and see the web ui.
Hope that can help someone.

akhepcat, 2021-11-05:
One issue with this is that there's apparently a change with numpy, and now the ai crashes on load.

There's already a bug open on it, so you might want to go give it a thumbs-up

isthisausername1, 2021-11-05:
I did see that in the logs but didn't think to much of it. That's not the only problem I'm having, whenever I try to put it into auto mode it keeps saying "[INFO] waiting for bettercap API to be available". However when I put it in my pi4 it works in auto fine. The passwords are still the default and I haven't changed them which is the common thing between all the reports of this problem on Google so I'm not sure if that's a me broke something thing Or a pi zero 2w thing.

Junohea, 2021-11-06:
> I think it defiantly could have been that, unfortunately i didn't check the version before updating but now after doing
> 
> sudo apt update && sudo apt full-upgrade -y
> 
> it started working and now it boots and i can ssh into it and see the web ui. Hope that can help someone.

Retried the 4 methods from before but did the dist-upgrade and it worked in all instances. 👍

isthisausername1, 2021-11-06:
> Retried the setups from before but did the full-upgrade and it worked in all instances. 👍

Glad to here it got it going, have you had any other problems running it on the pi zero 2 w

I manually started the better cap api now im getting ``error 400: exit status 254`` in the logs. Googleing this turned up nothing.

ianculpan, 2021-11-07:
> > I think it defiantly could have been that, unfortunately i didn't check the version before updating but now after doing
> > sudo apt update && sudo apt full-upgrade -y
> > it started working and now it boots and i can ssh into it and see the web ui. Hope that can help someone.
> 
> Retried the setups from before but did the full-upgrade and it worked in all instances. 👍

To clear up a little this sorted my install but I had to boot the image using a PI 3. The image wouldnt boot on a PI4. Once booted and upgraded everything looks to be working fine (so far)

skontrolle, 2021-11-10:
I've gotten the zero 2 to boot with the `apt full-upgrade` but haven't been able to get it to find any networks while running.  The `mon0` interface comes up and I can see networks with `iwlist mon0 scan`.  It gets to 50 blind epochs and reboots in 6 minutes.  
```
[CRITICAL] 50 epochs without visible access points -> rebooting ...
```

I notice this in the system log but doesn't seem to give many more clues.
```
brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2.txt failed with error -2
```

The same image works on the original 0w.

The AI issue I worked around by `apt install libavformat58 libavcodec58` but this replaced `kalipi-bootloader` with `raspberrypi-bootlader`.

akhepcat, 2021-11-10:
@skontrolle   you'll need to make sure you've got the most recent linux-firmware packages (libre and nonfree) installed.

Worst-case, you can grab the firmware files from any ubuntu machine running 21.04 or 21.10  right out of /lib/firmware/brcm/   and install them into the same place on the rPi

skontrolle, 2021-11-10:
Thanks, for the warning explanation.
The latest firmware appears to be in place and the checksums of the files match https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm 
`firmware-brcm80211/oldstable,now 1:20190114-2+rpt4 all [installed]` which was upgraded from rpt2.
If I remove the `brcmfmac43436-*` files the wifi doesn't come up at all as expected.

Is there something in bettercap I can set to log or try?

skontrolle, 2021-11-13:
I just retraced all of my steps with a clean `1.5.5` image to make sure I didn't have any issues with my backup config.

- raw image doesn't boot on zero2
- boot 0w
- `sudo apt update && sudo apt full-upgrade -y`
- swap card to zero2
- boots but no wifi for bettercap
  - install firmware not upgraded in the full-upgrade  `sudo apt install firmware-brcm80211`; reboot
- pwnagotchi starts, `mon0` interface up & can `iwlist mon0 scan`, numpy error in logs, all epochs are blind
  - update numpy `sudo pip3 install --upgrade numpy` to resolve #1015 
- with numpy fixed, all epochs are still blind, and the AI fails in `import cv2`
  - `sudo apt install libavcodec58 libavformat58`; these remove the kalipi-bootloader.  reboot for good measure
- epochs still blind, AI starts really quickly, pwnagotchi restarts after 50 blind epochs.
- drop card back in 0w, everything works.

akhepcat, 2021-11-15:
Okay, so @skontrolle - 
 i'm not sure if you'll be able to revert back, but you can try to force re- install  kalipi-bootloader and kalipi-kernel

i *think* that without the kalipi kernel (and modules) you won't get the correct driver for the mon interface, so won't be able to sniff correctly.

I was able to just install   libavcodec58 libavformat58  without upgrading/installing  the raspberrypi-bootloader/kernel by marking the two packages as "hold"  - apt-mark hold kalipi-bootloader   kalipi-kernel

it's a horrible workaround, but seems to have gotten mine up and running fully now ( also setting /etc/default/crda  to the correct domain)

skontrolle, 2021-11-15:
Thanks @akhepcat but still no joy on the wifi

The kalipi-kernel has persisted through the updates. I reverted the av..58s reinstalled the `kalipi-bootlaoder` held them but was unable to install the files.  Looking a bit more I found the `libraspberrypi0` is the reason the bootloader is tried to replace.  Installing a slightly older one allowed me to get the av..58 files installed and to keep the bootloader.
`libraspberrypi0/oldstable 1:1.20211029-1~buster armhf [upgradable from: 5.4.83-20210516]`

I also set the `/etc/default/crda`.  When messing manually in bettercap, I was able to see it failing on channels outside my locality with `wifi.recon on` so this might be doing something.  

For good measure, here's the wifi loading line from the kernel, is yours similar?
` brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct  9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270`

I'm not sure it's supported upstream in nexmon yet...
https://github.com/seemoo-lab/nexmon/issues/500

isthisausername2, 2021-11-15:
> * apt install libavcodec58 libavformat58

This worked perfectly, everything is working now and it is working great.

i made a quick script for anyone wanting to use the new rpi and is lazy like me. If you see anything i could do better please let me know, this is my first ever script.

https://github.com/isthisausername2/pwnagotchi_rpi_zero_2_fix

andur82, 2021-11-15:
> This worked perfectly, everything is working now and it is working great.
>
> i made a quick script for anyone wanting to use the new rpi and is lazy like me. If you see anything i could do better [...]

I'm sorry but... What exactly working great?
I've follow all steps from the first comment, and tried everything... But my rpi 02w still not working.
If I do "sudo iwlist mon0 scan" I receive all networks around me, but bettercap didn't found anything. It's me or nobody with the new zero have it working?

Btw I didn't update libav*58 because I didn't understand what to do after with kalipi-bootloader.

isthisausername2, 2021-11-15:
The steps written by @skontrolle got my Zero 2 w working. 

I recommend flashing your sd card with a fresh install of pwnagotchi, following those steps or using the script on a pi that does boot, then putting that fresh sd card in your zero 2 w and seeing if that fixes it. That's how i got mine working.

akhepcat, 2021-11-15:
I cloned the repo from @andur82   and made some changes to ward off some potential issues that I foresee.
https://github.com/akhepcat/pwnagotchi_rpi_zero_2_fix

@isthisausername2    burn a new, fresh image of 1.5.5.   
Boot onto a pi0w-v1 and either using a USB-ethernet or  the usb-gadget  interface for network connectivity,   run the script as root.  It'll take about an hour to download and upgrade.
You can't use wifi to do the upgrade, currently.

you should also be able to use a pi3 or pi4 for the first booting and upgrades (again, you'll need to use usb-gadget or ethernet for network connectivity)

That said:   it still won't work on the pi zero v2,  because the only firmware available for the brcmfmac43436  chip doesn't support the full capture,  unlike the  brcmfmac43430  of the -v1.

until  nexmon releases a working firmware, running this on the zero-v2  isn't going to get us anywhere.
https://github.com/seemoo-lab/nexmon/issues/500

andur82, 2021-11-15:
@akhepcat  many thanks for this explain.
I've done all stuff by hand and not by script on a pi3. Didn't understand about the nexmon firmware and the zero-v2. Sorry.

Strange things: on the rpi3, after done all update with libav*58 (and installing raspberry bootloader instead of the original kali) the pwnagotchi do only one round with the first found aps, then completely die. No more scan.
It's normal? Otherwise I'll back to older "original" 1.5.5 release.

skontrolle, 2021-11-16:
@isthisausername2 interesting that you got it to work.  What wifi firmware chipset /firmware do you get from `sudo dmesg |grep brcm`
I just tried a few of the nexmon patched pi3 firmwares without any luck. @akhepcat nice summary!

In a sense, the original issue is solved in getting the 0w2 to boot, but the wifi issue is a different blocker for at least my boards.

isthisausername2, 2021-11-16:
@akhepcat Thanks for that, i see some things added that i had no idea how to and also some other things that would help.

@skontrolle  I will try when i can. I remember seeing it catch a handshake but there are a few wifi networks around me so theres a chance it isn't actually working properly or they weren't in range. I took it on the bus with me today so i will see what or if its captured anything.
Should we make a new issue with an updated title or should we just leave it here and wait for the other softwares to be updated to support this new device.

skontrolle, 2021-11-16:
@isthisausername2 my initial hint that things weren't working was that the access points `CH * APS 0 (0)` always remained at 0.  The monitor interface will come up even if it can't actually monitor.

isthisausername2, 2021-11-16:
I think mine might have been saying that as well but i cant remember exactly, i will be able to check in around 2 hours and i can report back.

isthisausername2, 2021-11-16:
i was wrong unfortunately, i didn't check to see if it was actually working and just assumed that because the webui was working and it said looking for handshakes it was working properly.

davenicoll, 2021-11-16:
Once a firmware for the brcmfmac43436 that supports monitor mode is available, we'll be back in business. There's enough changes since the last image was built to warrant an updated image IMO, so I'll start work on that.

akhepcat, 2021-11-16:
> @akhepcat Thanks for that, i see some things added that i had no idea how to and also some other things that would help.

Sure - you want me to PR on your script so you can import the changes quickly?

isthisausername2, 2021-11-16:
> Once a firmware for the brcmfmac43436 that supports monitor mode is available, we'll be back in business. There's enough changes since the last image was built to warrant an updated image IMO, so I'll start work on PR for that.

Sounds good, please do let me know if you need any help beta testing the new image once the firmware is updated, i would he happy to help.

> Sure - i'll send a PR for your script so you can import the changes as you like, @isthisausername2 
Thank you, i will merge it and give you credit on the docs as well.

evo11x, 2021-11-17:
Same problem here with the Pi Z2 W, the image does not boot, the led is always off on the Pi and no display on the hdmi output. I have tried with Inky pHat and without with the same result.

akhepcat, 2021-11-17:
> Same problem here with the Pi Z2 W, the image does not boot, the led is always off on the Pi and no display on the hdmi output. I have tried with Inky pHat and without with the same result.

@evo11x  the wifi drivers do not currently work with the PiZ2.  Unless you're able to work on developing the appropriate nexmon wifi patches, following the script referenced above to allow you to boot on the Z2  won't get you very far.

matglas, 2022-01-15:
Some progress was just made here on nexmon https://github.com/seemoo-lab/nexmon/issues/500#issuecomment-1013734312 hopefully an update image can be created soon to make this work again.

isthisausername2, 2022-02-03:
We are very close to having this working on rpi 0 w 2
[https://github.com/seemoo-lab/nexmon/issues/500#issuecomment-1027755039](url)

dougdalton, 2022-02-07:
I have a 
[waveshare Raspberry Pi Zero 2 W  1GHz Quad-Core 64-bit Arm Cortex-A53 CPU, 512MB LPDDR2 SDRAM 2.4GHz 802.11 b/g/n Wireless LAN, WiFi Bluetooth 4.2 BLE]

I can boot :
Release date: January 28th 2022
System: 32-bit
Kernel version: 5.10
Debian version: 11 (bullseye)

but not the [v1.5.5] Image

I just get the rainbow screen,  how can I resolve this?

akhepcat, 2022-02-08:
> I just get the rainbow screen, how can I resolve this?

Wait until the next build release comes out with complete support for Zero2w.

BirbEXE, 2022-02-19:
New PR for nexmon is out

should it work ootb or will pwnagotchi need an update as well?

https://github.com/seemoo-lab/nexmon/issues/500#issuecomment-1046103826

moni11811, 2022-02-19:
> should it work ootb or will pwnagotchi need an update as well?
> 
> [seemoo-lab/nexmon#500 (comment)](https://github.com/seemoo-lab/nexmon/issues/500#issuecomment-1046103826)

I've currently tried it on pwnagotchi v1.5.5 with the script that patches the Pi2W (using Pi4) - not working... Wlan0 scans fine, mon0 seems to be scanning but no APs are showing up

moni11811, 2022-02-20:
I finally got it working
Confirmed pwnagotchi on RPI2W!!

[image]

I'll try to pull up a guide tomorrow but this was really hacky as you can tell from my previous comment

DrSchottky, 2022-02-20:
I made a pwnagotchi test image based on latest Raspberry Pi OS Lite Stable
https://we.tl/t-4jfxRWGXq2
Untested but should work
Still have [this bug](https://github.com/seemoo-lab/nexmon/issues/500#issuecomment-1046295728).

moni11811, 2022-02-20:
> I made a pwnagotchi test image based on latest Raspberry Pi OS Lite Stable https://we.tl/t-4jfxRWGXq2 Untested but should work Still have [this bug](https://github.com/seemoo-lab/nexmon/issues/500#issuecomment-1046295728).

Thank you for all your work - yes, I noticed that pwnagotchi just gets stuck in AUTO still does everything except switch to AI
`Expected 44 from C header, got 40 from PyObject`

AlexWhitehouse, 2022-02-21:
I've tested the image and can confirm it boots. Can anyone confirm whether their eInk (mine is WaveshareV2) display works on RPIw2?

moni11811, 2022-02-21:
> I've tested the image and can confirm it boots. Can anyone confirm whether their eInk (mine is WaveshareV2) display works on RPIw2?

I've got the waveshare_v2 BW works fine (although it's already damaged because of how many times I've removed it)

moni11811, 2022-02-21:
Looks like with the newest update by DrSchottky makes pwnagotchi officially working on barebone RPi02W 👯 

I've created a script working from DrSchottky's image: https://we.tl/t-4jfxRWGXq2
[scriptRPi02W.sh.txt](https://github.com/evilsocket/pwnagotchi/files/8111702/scriptRPi02W.sh.txt)

BirbEXE, 2022-02-22:
@evilsocket is there any chance of an official release that
supports the zero 2 w?

DrSchottky, 2022-02-22:
Unlike I did with test image the original image is based on kalipi kernel/drivers/fw, where my patches for 02w haven't been merged (yet).
Once it's done making a new release is trivial, you just have to change a bunch of commands in builder scripts.
AFAIK @evilsocket is taking a break so he probably won't update this anytime soon

danines, 2022-02-24:
> I've tested the image and can confirm it boots. Can anyone confirm whether their eInk (mine is WaveshareV2) display works on RPIw2?

I can confirm this works on my Zero2W with the waveshare V2 eink screen. 
Currently playing with the image provided from @DrSchottky .

XZCendence, 2022-02-25:
It seems like AI does not work with the zero 2 W? has anyone got it working successfully with either built in interface or external card (if so which one)

moni11811, 2022-02-25:
> It seems like AI does not work with the zero 2 W? has anyone got it working successfully with either built in interface or external card (if so which one)

It's really iffy. Works, then dies, then works for longer, then dies. This has been a common issue (since the plugin Watchdog exists) for a while it seems and the only known fix is to install 1.5.3 boot on a pi4, apply all updates and patches, then put it back in pi02w - from what Ive heard - moreover dont forget to expand your filesystem - it also causes issues

Otherwise the best option is to enable watchdog and itll automatically restart until it works

XZCendence, 2022-02-25:
Does it need to be a pi4? I have a 3b+ and a original zero W laying around.

moni11811, 2022-02-25:
> Does it need to be a pi4? I have a 3b+ and a original zero W laying around.

Sorry, in my case I used a Pi4 but it doesnt matter - anything that boots the old images should work.

XZCendence, 2022-02-25:
so if I boot an image on my 3b+, update and patch everything, and throw it back in my 02w then it should work? AI and everything else? I also want my pwnagotchi to be ever living haha, I want to make it as reliable as I can

moni11811, 2022-02-25:
Can you try changing

> so if I boot an image on my 3b+, update and patch everything, and throw it back in my 02w then it should work? AI and everything else? I also want my pwnagotchi to be ever living haha, I want to make it as reliable as I can

According to theories, yes, but I doubt it considering I went through that boat before DrSchottky posted his working image

moni11811, 2022-02-25:
Pwnagotchi as a whole is a finished product for the pi0w, waveshare_v2 and UPSlite otherwise anything else and you are playing wackamole

moni11811, 2022-02-25:
It seems numpy 1.21.5 is working

I just ran `pip3 install --upgrade numpy`

timmosaurusrex, 2022-02-27:
I am using @DrSchottky test image. My bettercap web-ui works just fine but my pwnagotchi ui is not working. Is that a bug? or did I do something wrong?

[103 more replies]