evilsocket/pwnagotchi #1048: create CI on github actions

peterus, 2021-11-12:
Hi @evilsocket !

I would like to contribute to pwnagotchi and do some improvements as I have a lot of knowledge in CI.
But first I would like to align here with you not that I am spending my time for nothing in the end.

I have seen that you are using Travis to build an image which is already very nice. But I think it would be better to create this images more often (every pull request and on the mainline), also to do some checks (code check, style check, are there dependency issues etc.). Testing on real hardware is simpler too as I can get a image directly from the github server and I do not need to setup something at home.
As Github actions is already very well integrated in Github, I propose to switch from Travis to Github Actions as it is simpler to use and it has nearly no restrictions.
The process to create a new release is also improved a lot if you have some q-gates already setuped.

To summerize my changes:
- switch from Travis to Github Action
- run CI (create image, code check, style check, ...) on every pull request and on the mainline -> result is directly shown on Github (green/red)
- add some more CI tasks

Please tell me what you are thinking about it. I think this will fix a lot of issues with dependencies (like numpy is not working in 1.5.5) and everybody will get a newer image anytime.

evilsocket, 2021-11-12:
I think this is the ideal solution, also to test for regressions when we, for instance, want to upgrade a package to a newer or simply different version (like in numpy's case).

incyi, 2022-07-19:
I tried this. But could not get it work. Anyone who knows why not ? 

Here is the branch:
https://github.com/incyi/pwnagotchi2/tree/github-actions

The repository folder changed warning is coming. But I can't suppress it. :

==> pwnagotchi: E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
78
==> pwnagotchi: E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
79
==> pwnagotchi: fuser -k /tmp/armimg-4210747451
80
Build 'pwnagotchi' errored after 42 seconds 806 milliseconds: Script exited with non-zero exit status: 100. Allowed exit codes are: [0]

incyi, 2022-07-25:
Oke, I think I found a solution for this issue what I saw. (Add this "--allow-releaseinfo-change" to apt update command) I will cleanup and start a pull request. But I am on vacation and can not test the created images now.