evilsocket/pwnagotchi #1057: system updates incorrectly prefer debian over kali repos [BUG]

akhepcat, 2021-11-17:
**Describe the bug**
When updating the system, the debian repositories are preferred over the kali-pi repositories,
causing breakage across different packages

**To Reproduce**
Steps to reproduce the behavior:
1. Install 1.5.5 image
2. apt --allow-releaseinfo-change update; apt -y full-upgrade
3. weep at the breakage

**Expected behavior**
the environment should set up the correct priorities for the kali-pi repositories to prevent breakage

**Environment (please complete the following information):**
 - Pwnagotchi ver  1.5.5  (and older, presumably)
 - Raspberry Pi

**Additional context**
This can be easily solved by performing the following:

```
cat <<EOF >/etc/apt/preferences.d/kali.pref
Package: *
Pin: release n=kali-pi
Pin-Priority: 999
EOF

```

This should be added to the default build, so that it prevents problems in the future, especially given long times between pwnagotchi releases.

akhepcat, 2021-11-17:
This fixes the upgrade issues when migrating from the  Pi-Zero-v1   to  Pi-Zero-v2

davenicoll, 2021-11-22:
Noted, thanks @akhepcat ... I'll include this in my work to update the pwnagotchi image.