evilsocket/pwnagotchi #1062: [BUG] bettercap/pwnagotchi crash on rPi Zero 2 W

alexl83, 2021-12-05:
**Describe the bug**

Using rPi Zero 2 W with external supported wifi device (in this case rt5592) in-kernel drivers 
pwnagotchi works for a variable time between 7 and 60 minutes, after that it crashes due to bettercap crashing
` (fatal error: concurrent map iteration and map write)`
pwnagotchi goes blind right after bettercap's crash until 50 blind epochs trigger a reboot

Both sanitized pwnagotchi and bettercap logs + 2.32 build attached

**To Reproduce**
Steps to reproduce the behavior:
1. connect external wifi (onboard disabled in config.txt)
2. power on raspberry pi 2 wireless
3. let it run 7 to 60 minutes
4. collect the crashlogs

**Expected behavior**
running without crashes

**Screenshots**
Please review attachments

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5 
 - OS version raspberry os buster fully upgraded (kali repo pinned) (kalipi packages held on 5.4.83-20210516)
 - Type of hardware Raspberry Pi Zero 2 Wireless
 - Any additional hardware used RT5592 wifi adapter

**Additional context**
- Same crashy behaviour on 2 different rPi zero 2 W
- tried pwnagotchi 1.5.5-shipped bettercap, tried official armhf 2.31.1 and natively compiled bettercap 2.32 from source using go1.17.4.linux-armv6l and official build instructions
- identical setup on rPi 0 W (first generation) doesn't lead to a crash (got 8+ hours uptime - all bettercap versions tried work without crashing) 

**Attachments**
[bettercap_crashlog.txt](https://github.com/evilsocket/pwnagotchi/files/7656230/bettercap_crashlog.txt)
[pwnagotchi_error_log.txt](https://github.com/evilsocket/pwnagotchi/files/7656229/pwnagotchi_error_log.txt)
[bettercap_2.32.zip](https://github.com/evilsocket/pwnagotchi/files/7656231/bettercap_2.32.zip)

AlexWhitehouse, 2021-12-09:
This is going to be related to #1046 . Nexmon needs an update (see issue https://github.com/seemoo-lab/nexmon/issues/500 ) for the RPi02w.

Brant2005, 2021-12-09:
这是来自QQ邮箱的自动回复邮件。你好，我可能不在，无法及时回复你的邮件。我将在看见邮件时，尽快给你回复。

alexl83, 2021-12-11:
> This is going to be related to #1046 . Nexmon needs an update (see issue https://github.com/seemoo-lab/nexmon/issues/500 ) for the RPi02w.

Are you sure Alex? I I'm not using nexmon at all, actually it's disabled in /boot/cmdline.txt and I'm using a ralink rt5592 adapter to monitor and inject packets

Thank you for your support!

Ale