evilsocket/pwnagotchi #1064: The quest to make AI mode work

scifijunk, 2021-12-11:
Well, this is my first post here so if I failed to follow the rules I'm sorry.

Now I'm fairly new when it comes to GitHub but I thought it would be cool to use the build script. With that being said I had to learn how to fix the problems with the script and finely got it to build. I then look to see how to get AI Mode to work by the means of removing and adding packages but that failed. Then I remembered seeing that evilsocket was not pleased with dadav not contributing to fixing a network problem with pwnagotchi. I went to dadav and looked at his __init__.py in his ai folder.

I note he changed (lines 48 and 49)

            a2c.load(config['path'], env)
            logging.debug("[ai] A2C loaded in %.2fs" % (time.time() - start))

to (lines 48 to 58)

            try:
                a2c.load(config['path'], env)
            except AssertionError as as_err:
                from fnmatch import fnmatch
                # Sometimes the model breaks...
                if not fnmatch(str(as_err), '* same * space as the model *'):
                    raise as_err
                else:
                    logging.warning("[ai] Model could not be loaded. Using new model.")
            else:
                logging.debug("[ai] A2C loaded in %.2fs" % (time.time() - start))

After I incorporated his changes I built my image and within about 4 or so minutes it went to AI Mode. I rebooted my test pwnagotchi and it went to AI Mode even faster say within 2 minutes. So maybe consider making a test pwnagotchi and see if it works on your end and if you like you can even try out my fork of pwnagotchi.

scifijunk, 2021-12-11:
Well it seems something else is broken now since it will not build again.

scifijunk, 2021-12-20:
Well, it seems that the repos that raspbian uses have been updated to include Python 3.8 or higher. I say that for the simple fact that my fork of pwnagotchi would not build until tonight (or this morning depending on how you want to look at it). I will burn on my SC Card to see if it works and let it run for 24 hours before making a pull request.

scifijunk, 2021-12-23:
Well, I was going to do a Pull Request but the papirus project has made a change to accommodate Debian 11 Bullseye and pwnagotchi with the current build scripts is aimed at Debian 10 Buster.

https://github.com/repaper/gratis/commit/87bf6b5494ee066f7961d4b83c01798244b44a4c#diff-95ea913fa6ba2aa215ec1080e782ae84705cd45986e3d8ca0cbf38cca978dcf4

As you can see they state:

"In Bullseye the bcm_host.h and libbcm_host.a are moved to the standard include and library directories. In Buster and older versions they were in a special directory tree /opt/vc.
Moving LDFLAGS to the end of the comman line makes it work on both Bullseye and Buster"

The problem is I don't think they did any testing to be sure that it would work for Debian 10 Buster.

With that being said I did get it to build before the papirus project made their change.

NUMAflex, 2022-09-27:
Just a heads up dont know if aware, but if you go 1.5.3 and allow it to self update to 1.5.5 the AI kicks in

scifijunk, 2022-09-28:
Well, I forgot to close this when I fixed the AI problem myself but thanks anyway.

closed 2022-09-28 by scifijunk