evilsocket/pwnagotchi #1068: [Feature Request] Waveshare UPS Hat (C) plugin

Waffelo, 2022-01-06:
It would be cool if someone with more skills could make a plugin for Waveshare UPS Hat that would display battery capacity and status (charging/discharging, voltage). I tried to make it by myself but I am too new to pyhton and I failed..

https://www.waveshare.com/product/ups-hat-c.htm

hannadiamond, 2022-02-05:
@Waffelo I just finished creating one! https://github.com/hannadiamond/pwnagotchi-plugins

makerph, 2022-02-27:
@hannadiamond tested it and works great. thanks

Waffelo, 2022-02-27:
@hannadiamond It seems to be working for me! I thank you so much! :D

closed 2022-09-10 by Waffelo

ch4istO, 2023-02-07:
hello, i am using the software:
https://github.com/DrSchottky/pwnagotchi/releases/tag/v1.5.6-beta2
I have the UPS_HAT C overlay, I copied the plugin and added it to config.toml but I can't activate it, how to solve the problem?

do-ki, 2023-02-09:
have you enable i2c?

ch4istO, 2023-02-09:
> have you enable i2c?

yes, but i solved the problem by installing pwnagotchi 1.5.5_WS V3 Patched, everything works in this software.