evilsocket/pwnagotchi #1071: No valid Handshakes even with aircrackonly plugin

Besix2, 2022-01-13:
Hello,
Today i set pwnagotchy up on a raspberry pi 4. After about 5 minutes it collected a few handshakes. I tried cracking these hashes with onlinehashcrack and distributed wpa cracking and both said no valid Handshakes so i deleted the handshakes and enabled aircrackonly plugin. Tried it again and still no valid handshake. What am i doing wrong? I even tried several tools like pwngetty and still no handshakes. Pls Help.

Thanks for your time 

P.S. sorry for spelling issues englisch is not my main language.

undermuz, 2022-02-15:
i also faced with this problem
but if `set wifi.handshakes.aggregate false; set wifi.handshakes.file ~/bettercap-wifi-handshakes.pcap` in bettercap, and send `bettercap-wifi-handshakes.pcap` to onlinehashcrack, it submitted success
seems like bettercap bug

Besix2, 2022-02-16:
Thanks for the answer i will try this as soon as possible and if its fixes the bug i Report it to bettercap.

undermuz, 2022-02-17:
@Besix2 
I've reported: https://github.com/bettercap/bettercap/issues/939

Besix2, 2022-08-27:
hello its me again,
I got it working one time than i got bored a tried something else but now I have installed pwngotchi again and I have the same problem I already checked if `set wifi.handshakes.aggregate true; set wifi.handshakes.file ~/bettercap-wifi-handshakes.pcap` is set and it is. I installed aircrack only plugin and tried 3 times but still no valid handshake.

Thanks