evilsocket/pwnagotchi #1072: Peer not working.

H4ck3r404, 2022-01-14:
My pwnagotchi can't detect other units and is not found on them (Which are working fine). Other aspects are working fine. Tried 1.5.5 but it couldn't enter AI mode.

Hardware - rpi0w v1.1
Software - pwnagotchi iso version - 1.5.3
Config - 

'''
main.name = "pwnagotchi"
main.whitelist = [

]
main.plugins.grid.enabled = false
main.plugins.grid.report = true
main.plugins.grid.exclude = [

]

main.plugins.onlinehashcrack.enabled = true
main.plugins.onlinehashcrack.email = ""
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false
main.plugins.onlinehashcrack.whitelist = []

main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 30
main.plugins.bt-tether.devices.android-phone.max_tries = 0
main.plugins.bt-tether.devices.android-phone.share_internet = true
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = false
main.plugins.bt-tether.devices.ios-phone.search_order = 2
main.plugins.bt-tether.devices.ios-phone.mac = ""
main.plugins.bt-tether.devices.ios-phone.ip = "172.20.10.6"
main.plugins.bt-tether.devices.ios-phone.netmask = 24
main.plugins.bt-tether.devices.ios-phone.interval = 5
main.plugins.bt-tether.devices.ios-phone.scantime = 20
main.plugins.bt-tether.devices.ios-phone.max_tries = 0
main.plugins.bt-tether.devices.ios-phone.share_internet = false
main.plugins.bt-tether.devices.ios-phone.priority = 999

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.net-pos.enabled = false
main.plugins.net-pos.api_key = "test"

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.webgpsmap.enabled = false

main.plugins.wpa-sec.enabled = false
main.plugins.wpa-sec.api_key = ""
main.plugins.wpa-sec.api_url = "https://wpa-sec.stanev.org"
main.plugins.wpa-sec.download_results = false
main.plugins.wpa-sec.whitelist = []

main.plugins.wigle.enabled = false
main.plugins.wigle.api_key = ""
main.plugins.wigle.whitelist = []
main.plugins.wigle.donate = true

[... 150 lines cut ...]
 "mod.started",
]

fs.memory.enabled = false
fs.memory.mounts.log.enabled = false
fs.memory.mounts.log.mount = "/var/log"
fs.memory.mounts.log.size = "50M"
fs.memory.mounts.log.sync = 60
fs.memory.mounts.log.zram = true
fs.memory.mounts.log.rsync = true

fs.memory.mounts.data.enabled = false
fs.memory.mounts.data.mount = "/var/tmp/pwnagotchi"
fs.memory.mounts.data.size = "10M"
fs.memory.mounts.data.sync = 3600
fs.memory.mounts.data.zram = false
fs.memory.mounts.data.rsync = true
'''

In /var/log/pwnagotchi.log was flooded with python3 exception handling errors.
Only thing i've seen with timestamp (others were loop of exception handling):

'''
[2022-01-13 12:37:17,614] [INFO] error 400: wifi.interface not set or not found  [2022-01-13 12:37:17,616] [ERROR] main loop exception (error 400: wifi.interface not set or not found)                                                            Traceback (most recent call last):                                                 File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/bettercap.py", line 11, in decode                                                                           return r.json()                                                                File "/usr/lib/python3/dist-packages/requests/models.py", line 897, in json        return complexjson.loads(self.text, **kwargs)                                  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads                                                                                    return _default_decoder.decode(s)                                              File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode                                                                                    obj, end = self.raw_decode(s)                                                  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode                                                                                return self.scan_once(s, idx=_w(s, idx).end())                               simplejson.errors.JSONDecodeError: Expecting value: line 1 column 1 (char 0)                                                                                      During handling of the above exception, another exception occurred:                                                                                               Traceback (most recent call last):                                                 File "/usr/local/bin/pwnagotchi", line 55, in do_auto_mode                         agent.recon()                                                                  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/agent.py", line 157, in recon                                                                               self.run('wifi.recon.channel clear')                                           File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/bettercap.py", line 55, in run                                                                              return decode(r, verbose_errors=verbose_errors)                                File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/bettercap.py", line 19, in decode                                                                           raise Exception(err)                                                         Exception: error 400: wifi.interface not set or not found                        [2022-01-13 12:37:17,771] [INFO] error 400: wifi.interface not set or not found  [2022-01-13 12:37:17,773] [ERROR] main loop exception (error 400: wifi.interface not set or not found)
'''

LamonJ, 2023-07-03:
Have you found a fix?

H4ck3r404, 2023-07-06:
looks like it might be hardware affected - rpi0w 1.1 always had some issues in other projects