evilsocket/pwnagotchi #1075: [BUG] New image (1.5.5) does not boot RPi Zero W 2

captbrando, 2022-02-08:
**Describe the bug**
Booting from fresh SD card gets stuck at the color-wheel screen when hooked up to HDMI. Verified the Pi will boot to standard Raspbian.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash pwnagotchi image
2. Add config.toml to /boot
3. Attempt to boot.

**Expected behavior**
Should boot and follow along with the configuration guidelines here: https://pwnagotchi.ai/configuration/

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - 1.5.5
 - Default for 1.5.5 image
 - Raspberry Pi Zero 2 Wireless
 - HDMI monitor to confirm state

**Additional context**
Does not appear to get past this critical spot. This is with or without the paper white screen, and no green LED light either.

captbrando, 2022-02-09:
Just some follow-on testing to make sure my hardware wasn't broken. Able to boot the image into Pi Zero 1 Wireless fine. All works as expected. Let the file system expand, blah blah, switched to the 2 and still stuck on the color wheel screen. Zero 2 boots with other images fine.

nathanjnorris, 2022-02-09:
Hey @captbrando - this is related to https://github.com/evilsocket/pwnagotchi/issues/1046. Nexmon needs an update - see issue https://github.com/seemoo-lab/nexmon/issues/500

captbrando, 2022-02-09:
Gotcha! I'll close this one and we can reference #1046

closed 2022-02-09 by captbrando