evilsocket/pwnagotchi #1076: [BUG] "decryption-webserver" not executable

Badb100d, 2022-02-09:
**Describe the bug**
When cryptobox is initialized, decryption webserver is not started when reboot.
Seems like the executable flag of the file  "/usr/bin/decryption-webserver" is not set.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 1.5.5 image and setup crypto configurations.
2. Reboot and connect to "Decrypt-me" wifi hotspot from another device.
3. Visit the pwnagotchi's IP address, seems connection refused, and cryptobox cannot get decrypted.

**Expected behavior**
When visiting the pwnagotchi's IP, it is supposed to be a webpage to get the password of cryptobox, but it's not after a fresh setup.

**Screenshots**
Where and how pwnlib do decryption
[image]

Script not executable because of non-executable flag.
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5 release image file

wyattearp, 2022-07-25:
Confirmed the above is the fix:

```
chmod +x /usr/bin/decryption-webserver
reboot
```

hybrezz54, 2023-12-29:
Confirmed this fix works without the need to reboot. Can this fix be added to the documentation (https://pwnagotchi.ai/configuration/), so it's more visible?