evilsocket/pwnagotchi #1079: [BUG] Hash does not match the Hash of the zip.

hubertron, 2022-02-15:
**Describe the bug**
Hash does not match the Hash of the zip.

**To Reproduce**
```
❯ shasum -a 256 pwnagotchi-raspbian-lite-v1.5.5.sha256
6b63fc3bead497e5a52dcc374327674d43409b904d1183011aee24a251ea19ff  pwnagotchi-raspbian-lite-v1.5.5.sha256
❯ shasum -a 256 pwnagotchi-raspbian-lite-v1.5.5.zip
608b7cc683ab60a4442b8b329332e7b923f29bea930f0e36e1eb2ff3a82f2b6b  pwnagotchi-raspbian-lite-v1.5.5.zip
❯ cat pwnagotchi-raspbian-lite-v1.5.5.sha256
4d4c0ad50f8bcac92827de42fc653a0f53045fc39866aae89e0e6cc42596f788  pwnagotchi-raspbian-lite-v1.5.5.img
```

**Expected behavior**
Hash's should match

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5

**Additional context**
File integrity looks ok but hashes don't match. Hesitant to build given this is the case.

hubertron, 2022-02-15:
I unzipped in VM and those hashes match.. Just not the zip. A malicious ZIP could exist though..

TavisMolyneux, 2022-03-10:
The hash of the .img inside the zip should match the text of the .sha256 file. Documentation does not make this clear for users unfamiliar with checksums but you dont want the hash of the .sha256 file as obviously that would not match the hash of a different file.