evilsocket/pwnagotchi #1080: [BUG] No Display on Waveshare screen

ZhuDaHai, 2022-02-18:
**Describe the bug**
No display on Raspberry Pi Zero W with Waveshare Rev2.1 screen. 

**To Reproduce**
Installed latest release v1.1.5, set up config.taml with the following display parameters:
```
ui.display.enabled = true
ui.display.type = "waveshare_2"
ui.display.color = "black"
```
No display. 
Tried: 
waveshare213d
waveshare_1
Still no display

**Expected behavior**
Should get a display, right?

**Screenshots**
This is my display. Pretty sure its a Waveshare v2 display
[image]

**Environment (please complete the following information):**
 - Pwnagotchi 1.1.5
 - OS version pwnagotchi-raspbian-lite-v1.5.5.img
 - Type of hardware Raspberry Pi Zero 0
 - Any additional hardware used none

**Additional context**
Boots fine. Can get into it via SSH over USB. 
Don't know where to look for any more diagnostics

HELP!

Thanks!

ZhuDaHai, 2022-02-18:
FYI: I saw pull request 1069 to add V3 support, but have No CLUE how to make those changes or even download them.
There is a comment in that conversion that 'detailed instructions' were left on redit on how to make the changes:

>  In the short term you can copy over the source changes.

Not much detail. 
Frustrated

ZhuDaHai, 2022-02-18:
Ok. I did what one of the commenters in that pull request did: downloaded the source and replaced the entire pwnagotchi folder.
Now it seems to work.

closed 2022-02-18 by ZhuDaHai

lcrince, 2022-06-20:
How did you build the install after. I seem to be having the same issue as you.