evilsocket/pwnagotchi #1082: Waveshare rev 2.1 not loading

SightsSet, 2022-02-23:
I've tried two different Waveshare displays, both rev 2.1 And I can't get either of them to display anything. I've tried Waveshare_2, _2.1, _v2, _v2.1, etc, commenting the line out entirely, nothing works. I also tried using jumpers to the header and still get nothing. I'm confident the GPIO header is soldered on fine. Log file doesn't show any hangups, just says initializing Waveshare and then moves on.

a8ksh4, 2022-02-23:
Is this relevant?
https://github.com/evilsocket/pwnagotchi/pull/1069

SightsSet, 2022-02-23:
Definitely, though I'm having trouble updating it. My python3.7 dir only has dist-packages, not site-packages and there is no pwnagotchi subdirectory in dist-packages.

SightsSet, 2022-02-23:
Yep. That worked! Thanks