evilsocket/pwnagotchi #1083: Accessing on Linux?

SightsSet, 2022-02-24:
Works fine SSHing via puTTY and the webui on windows, but I can't figure out how to get it configured for Linux.

thedroppedpocket, 2022-05-19:
webUI and SSH should work the same on linux and windows, What do you mean configured for linux?

Az-Neter, 2022-12-09:
By the tutorial on their own site, connecting ssh using linux simply fails.
There is no route setup by default and manually setting it does not work.
It seems like an oversight, but really it's just very bad code design. 
How SSH works on these little guys is inherently the worst thing I've ever experienced.
When you need to disable every single network interface on a preestablished network just to SSH into a device that literally already provides it's own capabilities to act as a router, something is very, very wrong.

virtadpt, 2022-12-09:
```
sudo sysctl net.ipv4.ip_forward=1
sudo iptables -t nat -A POSTROUTING -j MASQUERADE
sudo ifconfig enp0s20f0u1i1 10.0.0.1 netmask 255.255.255.0 up
sudo route add -net 10.0.0.0 netmask 255.255.255.0 dev enp0s20f0u1i1
```

Morpheus636, 2023-01-30:
@virtadpt 
Is this supposed to run on the pi or the device trying to connect to it?

virtadpt, 2023-01-30:
@Morpheus636 The device trying to connect to it.

b4dh0rse, 2023-06-05:
> ```
> sudo sysctl net.ipv4.ip_forward=1
> sudo iptables -t nat -A POSTROUTING -j MASQUERADE
> sudo ifconfig enp0s20f0u1i1 10.0.0.1 netmask 255.255.255.0 up
> sudo route add -net 10.0.0.0 netmask 255.255.255.0 dev enp0s20f0u1i1
> ```

For the next person who gets stuck on this on a modern Ubuntu system, the key is giving it the route (it's not in the original instructions).  10.0.0.0 and mask 255.255.255.0