evilsocket/pwnagotchi #1087: [BUG] error at pip -r reqiorments.txt

kocoman1, 2022-03-14:
<google-sheets-html-origin><style type="text/css"></style>

Using cached scipy-1.3.1.tar.gz (23.6 MB)
--
Installing build dependencies ... error
ERROR: Command errored out with exit status 1:
command: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-_1jmyw6w/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- wheel setuptools 'Cython>=0.29.2' 'numpy==1.13.3; python_version=='"'"'3.5'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.13.3; python_version=='"'"'3.6'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.14.5; python_version>='"'"'3.7'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.5'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.6'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version>='"'"'3.7'"'"' and platform_system=='"'"'AIX'"'"''
cwd: None
Complete output (6488 lines):
Ignoring numpy: markers 'python_version == "3.5" and platform_system != "AIX"' don't match your environment
Ignoring numpy: markers 'python_version == "3.6" and platform_system != "AIX"' don't match your environment
Ignoring numpy: markers 'python_version == "3.5" and platform_system == "AIX"' don't match your environment
Ignoring numpy: markers 'python_version == "3.6" and platform_system == "AIX"' don't match your environment
Ignoring numpy: markers 'python_version >= "3.7" and platform_system == "AIX"' don't match your environment
Collecting wheel
Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Collecting setuptools
Using cached setuptools-60.9.3-py3-none-any.whl (1.1 MB)
Collecting Cython>=0.29.2
Using cached Cython-0.29.28-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.manylinux_2_24_x86_64.whl (1.9 MB)
Collecting numpy==1.14.5

..

<google-sheets-html-origin><style type="text/css"></style>

\|                                             ^~~~~~~~~~~~~~~~~~~
--
numpy/random/mtrand/mtrand.c:45406:25: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
45406 \|                         (PyUnicode_GET_SIZE(**argname) != PyUnicode_GET_SIZE(key)) ? 1 :
\|                         ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
from /usr/include/python3.9/Python.h:97,
from numpy/random/mtrand/mtrand.c:4:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
446 \| static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
\|                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
error: Command "x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-1n6GrT/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-1n6GrT/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -Inumpy/core/include -Ibuild/src.linux-x86_64-3.9/numpy/core/include/numpy -Inumpy/core/src/private -Inumpy/core/src -Inumpy/core -Inumpy/core/src/npymath -Inumpy/core/src/multiarray -Inumpy/core/src/umath -Inumpy/core/src/npysort -I/usr/include/python3.9 -Ibuild/src.linux-x86_64-3.9/numpy/core/src/private -Ibuild/src.linux-x86_64-3.9/numpy/core/src/npymath -Ibuild/src.linux-x86_64-3.9/numpy/core/src/private -Ibuild/src.linux-x86_64-3.9/numpy/core/src/npymath -Ibuild/src.linux-x86_64-3.9/numpy/core/src/private -Ibuild/src.linux-x86_64-3.9/numpy/core/src/npymath -c numpy/random/mtrand/mtrand.c -o build/temp.linux-x86_64-3.9/numpy/random/mtrand/mtrand.o -MMD -MF build/temp.linux-x86_64-3.9/numpy/random/mtrand/mtrand.o.d" failed with exit status 1
----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-retxgtar/numpy_38093c62ebec4c80bf4bc98c1ed205c4/setup.py'"'"'; __file__='"'"'/tmp/pip-install-retxgtar/numpy_38093c62ebec4c80bf4bc98c1ed205c4/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-mhmjxvtt/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-_1jmyw6w/overlay --compile --install-headers /tmp/pip-build-env-_1jmyw6w/overlay/include/python3.9/numpy Check the logs for full command output.
----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/ee/5b/5afcd1c46f97b3c2ac3489dbc95d6ca28eacf8e3634e51f495da68d97f0f/scipy-1.3.1.tar.gz#sha256=2643cfb46d97b7797d1dbdb6f3c23fe3402904e3c90e6facfe6a9b98d808c1b5 (from https://pypi.org/simple/scipy/) (requires-python:>=3.5). Command errored out with exit status 1: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-_1jmyw6w/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- wheel setuptools 'Cython>=0.29.2' 'numpy==1.13.3; python_version=='"'"'3.5'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.13.3; python_version=='"'"'3.6'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.14.5; python_version>='"'"'3.7'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.5'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.6'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version>='"'"'3.7'"'"' and platform_system=='"'"'AIX'"'"'' Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement scipy==1.3.1
ERROR: No matching distribution found for scipy==1.3.1

KyleTryon, 2022-06-12:
I am here with the same issue. The project/repo seem dead and abandoned, I'm not sure we'll get this resolved without a fork.

LunarN0v4, 2023-11-04:
I'll have a look into it since I'm facing the same issue, although it seems to heavily rely on Python 3.7 (the last version SciPy 1.3.1 supports)