evilsocket/pwnagotchi #1091: bluetooth connexion

coutud, 2022-04-26:
hi i have probleme to connect my iphone with my pwnagotchi 
my iphone find it but the pwnagotchi stay at bt not found

thanks

Brets0150, 2022-08-06:
Did you try manually pairing to your phone from the pi?
https://pwnagotchi.ai/configuration/#bluetooth

coutud, 2022-08-07:
> Did you try manually pairing to your phone from the pi?
> 
> https://pwnagotchi.ai/configuration/#bluetooth

Yes i tried but my phone fond it but the pwnagotchi screen write nf

BEST regards

storm546, 2022-08-10:
same issue here but with android trying to connect manually after pairing outputs an error:
`Failed to connect: org.bluez.Error.Failed`
Does anyone have a fix for this?

chunkywill, 2023-02-02:
i had a similar thing when i tried connecting but i did remove <mac address> and then try manually pair it again and then it worked

chunkywill, 2023-02-02:
> same issue here but with android trying to connect manually after pairing outputs an error: `Failed to connect: org.bluez.Error.Failed` Does anyone have a fix for this?

i had a similar problem but did the comman remove  _your phones mac adress_
and then manually re paired it and it worked fine

waterfallm, 2023-03-13:
@chunkywill can you confirm your exact actions here as i've tried similar without much success

Pejda, 2023-03-19:
> @chunkywill can you confirm your exact actions here as i've tried similar without much success

I had similar problem and somehow believe it was solved thanks to updates.
I did "sudo apt update -y" and "sudo apt upgrade -y" and then "sudo apt dist-upgrade -y" 
After reboot i just did "sudo bluetoothctl remove [MAC]" - replace [MAC] with mac address of already paired device - if you did the pairing before updates.
Before updating my device, it was impossible to fix this for me.

waterfallm, 2023-03-20:
> I had similar problem and somehow believe it was solved thanks to updates.
> I did "sudo apt update -y" and "sudo apt upgrade -y" and then "sudo apt dist-upgrade -y"
> After reboot i just did "sudo [...]

Same after update altough i'm running witn version 1.6.2 no updates applpied phone does get pinged and paired but pwnagotchi always shows "NF"

Do i need anything else in my bluetooth config? will wrong tethering IP cause this?