evilsocket/pwnagotchi #1092: [BUG] empty bettercap events + warning: "http://pwnagotchi.local:8081/api is using an insecure connection, refer to the documentation to configure the api.rest module to use SSL."

rdavydov, 2022-04-26:
Quote from https://www.reddit.com/r/pwnagotchi/comments/jrqelv/issue_with_bettercap_web_ui/ : 

> When I log into bettercap it's not showing me any information in the event tab. It says the script is running and the API polling time is changing. In the advanced tab it says event.stream is running. All settings are default and The pwnagotchi is in manual mode and has several handshakes on it. If run sudo pwnagotchi while connected to Bettercap I can see the pwnagotchi working and the tasks it's performing in the terminal, but nothing on the event tab.
> 
> Also, not sure if it's related but Bettercap has a warning message that the connection is insecure and to change the api.rest module to use SSL. I've searched through the documentation and google but haven't found a solution.

I have the same issue.

luqezr, 2023-05-25:
Any updates on this? I have a similar issue

Boshimanu, 2023-07-10:
Same Problem here... thanks!

CallMeMaybe420, 2023-12-20:
same :D