evilsocket/pwnagotchi #1101: [BUG] AI Won't start on pi 3B+ (kali pi)

XZCendence, 2022-06-10:
Upon installing pwnagotchi on top of a kali arm OS install on a pi 3B+, pwnagotchi seems to run just fine until it tries to enable AI, It seems to get into ai "mode" as shown on the bottom right corner of the screen, but no brain.nn file is written and it throws this error: 
`[ERROR] [ai] error while training (could not broadcast input array from shape (428) into shape (1,503))
Traceback (most recent call last):
ValueError: could not broadcast input array from shape (428) into shape (1,503)`
**To Reproduce**

Steps to reproduce the behavior:
1. Install pwnagotchi on kali pi 
2. use wifi adapter with 5GHZ capability

**Expected behavior**
AI enables normally, and a brain.nn file is written to /root/

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5
 - OS version Kali ARM 2022.2
 - Type of hardware Pi 3 B+
 - RTL8811AU based wifi card (same behavior with MT7610 based card)

XZCendence, 2022-06-10:
I've jumped through many dependency and package hoops to get here, and I'm so close because this seems to be an identical issue to back when 5GHz channels were not supported past 140, which seems to have been fixed, but I'm still getting the issue on 1.5.5.
I've tried installing 1.5.3 and letting it update to 1.5.5 to no avail.

databonanza, 2022-06-16:
AI won't start on my unit either, the issue is not specific to the hardware you've got.

`#  pip3 install --upgrade numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/numpy/
Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='pypi.org', port=443): Read timed out. (read timeout=15)")': /simple/numpy/
^COperation cancelled by user
`

`# cat /etc/resolv.conf
# Generated by resolvconf
nameserver 9.9.9.9
nameserver 127.0.0.1`

`[2019-07-10 01:30:25,018] [ERROR] error while starting AI (numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/__init__.py", line 21, in load
    from stable_baselines import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/__init__.py", line 1, in <module>
    from stable_baselines.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/__init__.py", line 1, in <module>
    from stable_baselines.a2c.a2c import A2C
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 6, in <module>
    import tensorflow as tf
  File "/usr/lib/python3/dist-packages/tensorflow/__init__.py", line 98, in <module>
    from tensorflow_core import *
  File "/usr/lib/python3/dist-packages/tensorflow_core/__init__.py", line 35, in <module>
    from tensorflow._api.v1 import compat
  File "/usr/lib/python3/dist-packages/tensorflow_core/_api/v1/compat/__init__.py", line 21, in <module>
    from tensorflow._api.v1.compat import v1
  File "/usr/lib/python3/dist-packages/tensorflow_core/_api/v1/compat/v1/__init__.py", line 652, in <module>
    from tensorflow_estimator.python.estimator.api._v1 import estimator
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/__init__.py", line 8, in <module>
    from tensorflow_estimator._api.v1 import estimator
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/_api/v1/estimator/__init__.py", line 8, in <module>
    from tensorflow_estimator._api.v1.estimator import experimental
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/_api/v1/estimator/experimental/__init__.py", line 8, in <module>
    from tensorflow_estimator.python.estimator.canned.dnn import dnn_logit_fn_builder
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/__init__.py", line 25, in <module>
    import tensorflow_estimator.python.estimator.estimator_lib
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/estimator_lib.py", line 53, in <module>
    from tensorflow_estimator.python.estimator.inputs import inputs
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/inputs.py", line 22, in <module>
    from tensorflow_estimator.python.estimator.inputs.numpy_io import numpy_input_fn
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/numpy_io.py", line 26, in <module>
    from tensorflow_estimator.python.estimator.inputs.queues import feeding_functions
  File "/usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/inputs/queues/feeding_functions.py", line 40, in <module>
    import pandas as pd
  File "/usr/local/lib/python3.7/dist-packages/pandas/__init__.py", line 29, in <module>
    from pandas._libs import hashtable as _hashtable, lib as _lib, tslib as _tslib
  File "/usr/local/lib/python3.7/dist-packages/pandas/_libs/__init__.py", line 13, in <module>
    from pandas._libs.interval import Interval
  File "pandas/_libs/interval.pyx", line 1, in init pandas._libs.interval
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject
[2019-07-10 01:30:25,304] [WARNING] [ai] AI not loaded!`

Spessmen14, 2022-10-14:
Same error on my Pi0. Something about 1.5.5 broke the AI functionality.

ifeelagood, 2022-12-14:
same issue here, on PI3B+. First tried on the raspbian image 1.5.5 with an updated numpy, as well as 1.5.3.

Is it due to the 5GHz channel support on the PI 3B+?  I am going to try disabling 5Ghz support and see if that does anything.

Hopefully @evilsocket can shed some light on this issue. If not, i will take a look myself and see if I can do anything.