evilsocket/pwnagotchi #1105: [BUG] Pi4 4GB does not boot in MANU mode using ethernet and DHCP not providing IP

OfficialMuffin, 2022-07-21:
**Describe the bug**
Raspberry Pi 4 4GB does not boot in MANU mode when plugged in using ethernet and DHCP does not give assign an IP address to its ethernet interface.

**To Reproduce**
Steps to reproduce the behavior:
1. Use regular install instructions for Pi0w  
2. Plug ethernet cable into computer and Pi 4 4GB
3. Wait for boot

**Expected behavior**
Should boot in MANU mode when ethernet is detected and DHCP to provide an IP address for the interface.  

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.5.5
 - Debian 10
 - Raspberry PI 4 4GB

lithiumoxide, 2022-08-27:
I see similar behaviour with version 1.5.6-beta2 (1.5.5 won't boot at all for me) on a Pi0w. I connect the Pwnagotchi to my PC's USB port, on the RPi's data port, and the device boots into AUTO mode, eventually switching to AI mode.

llamasoft, 2022-09-16:
It looks like the way it detects if it should boot into AUTO/MANU mode is based on if usb0 (for Pi0W) or eth0 (for everything else) is up when pwnagotchi launches: 

https://github.com/evilsocket/pwnagotchi/blob/544314b88cee7f2ee488b49bd61636d664a4c7fa/builder/data/usr/bin/pwnagotchi-launcher#L15-L19  
https://github.com/evilsocket/pwnagotchi/blob/544314b88cee7f2ee488b49bd61636d664a4c7fa/builder/data/usr/bin/pwnlib#L53-L80

If you manage to log into your devices while they're in AUTO mode, can you run the following command and post the output?

```
for device in usb0 eth0; do echo "=== $device ==="; cat /sys/class/net/$device/operstate; echo; done
```