evilsocket/pwnagotchi #1109: waveshare 2.7 inch v3 screen support

cyanide-hacker, 2022-07-31:
waveshare 2.7 inch v3 screen is not working. If I preform the following:

teps:

Build a image from last version.

Do the first boot in your pwnagotchi with V3 display connected (they will not display anything).
Wait some minutes and SSH into your pwnagotchi.
In your host machine run:

- git clone https://github.com/ikornaselur/pwnagotchi.git
- cd pwnagotchi
- git checkout waveshare-v3-support
- zip -r ../pwnagotchi_scripts_folder.zip pwnagotchi/
- cd ..

Now you have pwnagotchi scripts folder ziped with v3 support, you just need to put this file inside your pwnagotchi, i use python to do that:
in your host machine in folder that have your zip file: python3 -m http.server 5000

- In your pwnagotchi: wget 10.0.0.1:5000/pwnagotchi_scripts_folder.zip
- Now stop pwnagotchi service sudo service pwnagotchi stop
- unzip the folder unzip pwnagotchi_scripts_folder.zip
- Delete old pwnagotchi scripts folder sudo rm -rf /usr/local/lib/python3.7/dist-packages/pwnagotchi
- Move the v3 support folder to right directory: sudo mv pwnagotchi_scripts_folder /usr/local/lib/python3.7/dist-packages/pwnagotchi
- Edit config file sudo nano /etc/pwnagotchi/config.toml
- change 'waveshare_2' to 'waveshare_3'
- save the file
- sudo service pwnagotchi start

I can get the screen to display, but it is very washed out and doesn't have much detail.

## Possible Solution
Can you please add support for the Waveshare v3 screens since the v2 are being phased out?

## Your Environment

- [X ] You're using the official images

- [X ] You're using a raspberry pi 0

- [ ] You're using a supported LCD

tdauria1, 2022-09-08:
> Move the v3 support folder to right directory: sudo mv pwnagotchi_scripts_folder /usr/local/lib/python3.7/dist-packages/pwnagotchi

To anyone else following this post line-by-line, the `pwnagotchi_scripts_folder.zip` extracts to `~/pwnagotchi/`. So the command from the quoted step would be `sudo mv pwnagotchi /usr/local/lib/python3.7/dist-packages/pwnagotchi`

Otherwise, this works as advertised and brought my v3 screen online, so thanks! Now I'm on to hunt down the fix for the undersaturated display..

tigabyte101, 2022-12-08:
Sorry if I am missing something here, but when running the command "git checkout waveshare-v3-support" from the "pwnagotchi" folder, I am getting the error "error: pathspec 'waveshare-v3-support' did not match any file(s) known to git" and the only branch listed is "master". What could I be missing?

tehtex, 2022-12-14:
i am also getting "error: pathspec 'waveshare-v3-support' did not match any file(s) known to git", i explored the entire clone. im afraid to admit that I have actually never used the "checkout" with git, so i'll research on that. but i saw i wasn't the only one with this error so thought i would join in,

user171, 2023-01-02:
the `checkout` command switches branch. And indeed it you look at the branches available on the linked repo, there is no branch named `waveshare-v3-support` at the moment.

DooMFeaR, 2023-03-28:
Any updates on this?

glowfishDE, 2023-08-03:
*ping*
I also was not able to get hands on a V2 Display anymore. Waveshare seems to only sell V3 displays recently. Can you please look into this and bring V3 support to the project? Thank you so much 👍