evilsocket/pwnagotchi #1110: [BUG] USB Wifi Dies around 45-60 minutes (RPI 4; Pwn 1.55)

imthisguymike, 2022-08-02:
**Describe the bug**
USB Wifi Dies around 45-60 minutes (RPI 4; Pwn 1.55) - Basically using 3 different USB wifi adapters (since the built-in RPI 4 wireless doesn't work). All 3 initially work fine, but then "die".

**To Reproduce**
Steps to reproduce the behavior:
1. Run for about 45-60 minutes, then pwnagotchi does not see any wireless networks and gets bored.
2. To resolve, reboot and it is fine.

**Expected behavior**
Expected that it will continue to work without needing a reboot after 45-60 minutes.

**Environment (please complete the following information):**
 - Pwnagotchi version - 1.55
 - Type of hardware - RPI 4
 - Any additional hardware used - Panda Wireless® PAU0A AC600, Realtek RTL8811AU (WN-H2), BrosTrend 650Mbps.

lucidyan, 2022-08-03:
I trying to solve same problem for myself, but I have some mix of different errors, according to `/var/log/pwnagotchi.log` logs. Can you attach yours?

imthisguymike, 2022-08-03:
When pwnagotchi is mad and it's not showing APs:
```
[2022-08-03 19:23:21,391] [INFO] [ai] saving model to /root/brain.nn ...
[2022-08-03 19:23:21,459] [INFO] [ai] saving /root/brain.json
[2022-08-03 19:23:21,517] [INFO] [ai] setting new policy:
[2022-08-03 19:23:21,517] [INFO] [ai] ! min_rssi: -152 -> -97
[2022-08-03 19:23:21,517] [INFO] [ai] ! ap_ttl: 166 -> 92
[2022-08-03 19:23:21,517] [INFO] [ai] ! sta_ttl: 225 -> 153
[2022-08-03 19:23:21,517] [INFO] [ai] ! recon_time: 56 -> 7
[2022-08-03 19:23:21,518] [INFO] [ai] ! min_recon_time: 23 -> 17
[2022-08-03 19:23:21,518] [INFO] [ai] ! max_interactions: 19 -> 14
[2022-08-03 19:23:21,518] [INFO] [ai] ! max_misses_for_recon: 3 -> 10
[2022-08-03 19:23:21,518] [INFO] [ai] ! excited_num_epochs: 8 -> 17
[2022-08-03 19:23:21,518] [INFO] [ai] ! bored_num_epochs: 21 -> 7
[2022-08-03 19:23:21,518] [INFO] [ai] ! sad_num_epochs: 16 -> 9
[2022-08-03 19:23:21,519] [INFO] [ai] ! channels: [4, 6, 7, 8, 10, 11, 13, 40, 44, 48, 64, 100, 104, 124, 132, 140] -> [1, 3, 8, 9, 12, 36, 48, 52, 104, 112, 116, 120, 124, 136, 140]
[2022-08-03 19:23:39,286] [INFO] [epoch 899] duration=00:00:17 slept_for=00:00:07 blind=17 sad=11 bored=0 inactive=32 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=0% mem=20% temperature=43C reward=-0.01$
[2022-08-03 19:23:39,287] [WARNING] 32 epochs with no activity -> angry
[2022-08-03 19:23:39,287] [INFO] [ai] --- training epoch 25/50 ---
[2022-08-03 19:23:39,290] [INFO] [ai] REWARD: -0.015222
[2022-08-03 19:23:39,290] [INFO] [ai] observation:
[2022-08-03 19:23:39,291] [INFO]     aps
[2022-08-03 19:23:39,291] [INFO]     sta
[2022-08-03 19:23:39,291] [INFO]     peers
```

After I reboot:
```
[2022-08-03 19:34:37,745] [INFO] [ai] saving model to /root/brain.nn ...
[2022-08-03 19:34:38,262] [INFO] [ai] saving /root/brain.json
[2022-08-03 19:34:45,214] [INFO] [ai] setting new policy:
[2022-08-03 19:34:45,214] [INFO] [ai] ! min_rssi: -196 -> -58
[2022-08-03 19:34:45,214] [INFO] [ai] ! ap_ttl: 490 -> 375
[2022-08-03 19:34:45,215] [INFO] [ai] ! sta_ttl: 79 -> 200
[2022-08-03 19:34:45,215] [INFO] [ai] ! recon_time: 51 -> 29
[2022-08-03 19:34:45,215] [INFO] [ai] ! max_inactive_scale: 4 -> 7
[2022-08-03 19:34:45,215] [INFO] [ai] ! recon_inactive_multiplier: 3 -> 1
[2022-08-03 19:34:45,215] [INFO] [ai] ! hop_recon_time: 58 -> 7
[2022-08-03 19:34:45,215] [INFO] [ai] ! min_recon_time: 20 -> 5
[2022-08-03 19:34:45,216] [INFO] [ai] ! max_interactions: 12 -> 16
[2022-08-03 19:34:45,216] [INFO] [ai] ! max_misses_for_recon: 3 -> 5
[2022-08-03 19:34:45,216] [INFO] [ai] ! excited_num_epochs: 28 -> 23
[2022-08-03 19:34:45,218] [INFO] [ai] ! bored_num_epochs: 25 -> 5
[2022-08-03 19:34:45,219] [INFO] [ai] ! sad_num_epochs: 7 -> 19
[2022-08-03 19:34:45,219] [INFO] [ai] ! channels: [1, 3, 4, 5, 6, 9, 10, 13, 36, 56, 60, 104, 116, 136] -> [2, 4, 5, 6, 8, 10, 13, 36, 52, 64, 104, 108, 116, 124, 128, 136, 140]
[2022-08-03 19:35:29,256] [INFO] sending association frame to UniversityHill_Unit (54:ec:2f:a8:16:f8 Ruckus Wireless) on channel 6 [0 clients], -44 dBm...
[2022-08-03 19:35:29,293] [INFO] sending association frame to UniversityHill_Unit (20:58:69:88:fd:38 Ruckus Wireless) on channel 6 [0 clients], -56 dBm...
[2022-08-03 19:35:29,333] [INFO] sending association frame to DIRECT-E6-HP ENVY 5540 series (fc:3f:db:c6:e0:e7 Hewlett Packard) on channel 6 [0 clients], -71 dBm...
[2022-08-03 19:35:29,379] [INFO] sending association frame to UniversityHill_Staff (20:58:69:48:fd:38 Ruckus Wireless) on channel 6 [0 clients], -56 dBm...
[2022-08-03 19:35:29,441] [INFO] sending association frame to TG1672G82 (98:6b:3d:63:82:80 ARRIS Group, Inc.) on channel 6 [0 clients], -74 dBm...
[2022-08-03 19:35:29,484] [INFO] sending association frame to UniversityHill_Staff (54:ec:2f:68:16:f8 Ruckus Wireless) on channel 6 [0 clients], -44 dBm...
[2022-08-03 19:35:29,505] [INFO] waiting for 5s on channel 6 ...
[2022-08-03 19:35:34,639] [INFO] CHANNEL 36
[2022-08-03 19:35:34,652] [INFO] 3 access points on channel 36
[2022-08-03 19:35:34,705] [INFO] sending association frame to UniversityHill_Staff (20:58:69:48:fb:1c Ruckus Wireless) on channel 36 [0 clients], -81 dBm...
[2022-08-03 19:35:34,789] [INFO] sending association frame to UniversityHill_Staff (54:ec:2f:68:16:fc Ruckus Wireless) on channel 36 [0 clients], -66 dBm...
[2022-08-03 19:35:34,876] [INFO] sending association frame to UniversityHill_Unit (54:ec:2f:a8:16:fc Ruckus Wireless) on channel 36 [0 clients], -66 dBm...
[2022-08-03 19:35:35,022] [INFO] [epoch 3] duration=00:00:57 slept_for=00:00:56 blind=0 sad=0 bored=0 inactive=0 active=3 peers=0 tot_bond=0.00 avg_bond=0.00 hops=2 missed=0 deauths=0 assocs=9 handshakes=0 cpu=4% mem=20% temperature=45C reward=0.15142857$
[2022-08-03 19:35:35,024] [INFO] [ai] --- training epoch 2/50 ---
[2022-08-03 19:35:35,027] [INFO] [ai] REWARD: 0.151429
[2022-08-03 19:35:35,030] [INFO] [ai] observation:
[2022-08-03 19:35:35,030] [INFO]     aps
[2022-08-03 19:35:35,031] [INFO]       CH 1: 0.14814814814759947
[2022-08-03 19:35:35,033] [INFO]       CH 6: 0.2222222222213992
[2022-08-03 19:35:35,034] [INFO]       CH 11: 0.2222222222213992
[2022-08-03 19:35:35,034] [INFO]       CH 36: 0.1111111111106996
[2022-08-03 19:35:35,035] [INFO]       CH 44: 0.2222222222213992
[2022-08-03 19:35:35,035] [INFO]       CH 48: 0.07407407407379973
[2022-08-03 19:35:35,036] [INFO]     sta
[2022-08-03 19:35:35,036] [INFO]       CH 11: 0.9999999999
[2022-08-03 19:35:35,037] [INFO]     peers
```

lucidyan, 2022-08-03:
Hmm, very clean logs (but unfortunately not complete, it is also recommended to erase MAC addresses and Wi-Fi names with RegExp). In my case, the Watchdog module reboots the system when it receives a bling-bug where it can't switch a channel due to a blind-bug in the nextmon drivers.

Also, maybe this is normal behaviour and there aren't too many networks around you?