evilsocket/pwnagotchi #1111: PiSugar 3

willproctor, 2022-08-07:
**Is your feature request related to a problem? Please describe.**
N/A

**Describe the solution you'd like**
Documented battery module update

**Describe alternatives you've considered**
N/A

**Additional context**
The PiSugar 3 includes an RTC on the expansion board. Adding this to the documentation would be useful for people building a pwnagotchi for the first time as it would mean that a seperate RTC module does not need to be soldered.

Ra2-IFV, 2022-09-20:
https://github.com/kellertk/pwnagotchi-plugin-pisugar2
https://github.com/taiyonemo/pwnagotchi-plugin-pisugar3
: )

D337z, 2023-04-15:
It's in the PiSugar3 website.  If you don't know what you're buying, why would you buy it?  There's no need to include specifics about hardware products other than what's absolutely recommended in software descriptions.  It should probably be more of a word-of-mouth thing to suggest for people building the PWNagotchi, or just leave it in the plugins which are in a completely DIFFERENT forum.

willproctor, 2023-04-15:
> It's in the PiSugar3 website.  If you don't know what you're buying, why would you buy it?  There's no need to include specifics about hardware products other than what's absolutely recommended in [...]

This reduces the thickness of the pwnagotchi and removes a hardware dependency. 

It works great and does not cause an issue. This issue is a suggestion for documentation being added to the repo per the contributing guidelines. 

If you would like to document this you are welcome to make a pull request and cite this issue.