evilsocket/pwnagotchi #1115: [BUG] pwnagotchi image can't be built

llamasoft, 2022-09-16:
**Describe the bug**
It's currently impossible to build a pwnagotchi image due to a number of reasons:

- The image's main apt repo changed from "stable" to "oldstable" which causes the initial `apt-get update` to fail.
- The re4son kernel apt repository keys are expired.
- The arm-image packer plugin repo has been renamed.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `make image`

**Additional context**
I recently started working on building a pwnagotchi but got a Waveshare v3 display.  I saw that support for the display had been merged but that no pwnagotchi images have been released in over a year.  I tried to build the image myself but kept running into issues.

I'm currently working on a fix for most of these. 👍

TQMatvey, 2022-09-19:
just tried to build using your fork and makefix branch, 
`cd builder && sudo /usr/bin/unshare --uts /tmp/packer build -var "pwn_hostname=pwnagotchi" -var "pwn_version=1.5.5" pwnagotchi.json
 Error loading configuration: 

fork/exec /tmp/packer-plugin-3560001340.zip: permission denied
make: *** [Makefile:56: pwnagotchi-raspios-lite-1.5.5.img] Error 1`

Are you also stuck on that?

llamasoft, 2022-09-19:
I haven't quite worked out what causes that issue yet, but if you rerun the makefile with sudo (i.e. `sudo make image`) it'll fix it.  
Ideally, the fix would be upgrading builder/pwnagotchi.json to HCL so it can use the [`required_plugins` directive](https://www.packer.io/docs/templates/hcl_templates/blocks/packer#specifying-plugin-requirements) instead of installing plugins manually.  That said, I was trying to keep my changes minimal and the JSON -> HCL conversion felt out of scope for just getting the builds working again.

TQMatvey, 2022-09-20:
anyways i seemingly fixed everything.. building now, going without any issues, fixed re4son keys also, will push my work to github a bit later

TQMatvey, 2022-09-20:
anyways i can confirm, i fully fixed builds and kernel keys, you can check my fork

Allstreamer, 2022-09-23:
Tryed Building with your fork, i get a warning and it fails:
```
Warning: Warning when preparing build: "pwnagotchi"

last_partition_extra_size is deprecated, use target_image_size to grow your
image
```

here is the failure:
```
==> pwnagotchi: losetup --show -f -P output-pwnagotchi/image
==> pwnagotchi: error losetup --show -f -P exit status 1: losetup: output-pwnagotchi/image: failed to set up loop device: No such device or address
==> pwnagotchi:
Build 'pwnagotchi' errored after 1 minute 8 seconds: step canceled or halted
```

D337z, 2023-04-15:
The issues are mainly that the versions of programs like Go, Packer and the like are so outdated that the newer programs that depend on them can't be built using them.  You have to manually download newer versions from the websites via wget and install them in order to use them.  On top of that, it seems that Packer was deliberately sabotaged as it includes scripts in the JSON that require sudo to run but won't work unless a separate bash script is used to run the commands.
It's also trying to pull an older 2019 version of Raspbian instead of trying to always pull the most recent version to create the image file with.
Oh, and good luck trying to compile a 32-bit version because you'll have to track down a specific version of a program (that I can't recall the name of at the moment) that works in 32-bit.  You'll also have to change the Makefile to download the correct items or just do it manually and erase those horrible bits of code.  And then there's the json file...just don't.  If you can take the bin files that you compiled and toss them into a working image, just do it and run far far away.  The JSON file will be a reference for you that Packer would otherwise require.
So, have fun!