evilsocket/pwnagotchi #1117: [BUG] Boot hangs at "Started Update UTMP about System Runlevel Changes" when Pi0 is installed in a PoE/ETH/USB HUB HAT

Spessmen14, 2022-09-23:
**Describe the bug**
After flashing a new image to a MicroSD card, when properly installed in the Waveshare PoE/ETH/USB HUB HAT, the boot stalls at the step "Started Update UTMP about System Runlevel Changes" and never completes.

https://www.waveshare.com/wiki/PoE/ETH/USB_HUB_HAT
**To Reproduce**
Steps to reproduce the behavior:
1. Flash new pwnagotchi image (have tried official and the prepatched WaveshareV3 version from here: https://archive.org/details/pwnagotchi_1.5.5_WSV3Patched) to a Micro SD car
2. Boot by connecting MicroUsb cable to DATA port on the HAT and a HDMI cord to the HDMI plug
3. System hangs at the UMTP prompt

**Expected behavior**
Boot to Pwn welcome info and console

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5
 - OS version : Windows 7, 64 bit
 - Type of hardware - Original Pi 0 
 - Any additional hardware used: https://www.waveshare.com/wiki/PoE/ETH/USB_HUB_HAT

**Additional context**
I have tried with an old MicroSD card and a brand-new, Industrial-grade MicroSD card from Kingston. Same result.

strange-world, 2023-07-04:
I am also experiencing this issue where boot will hang at 'Started Update UTMP about System Runlevel Changes.'

Environment
pwnagotchi-raspbian-lite-v1.5.5.img
OS version : Windows 11, 64 bit
Type of hardware - Original Pi 0 W
Micro SD - SanDisk 32GB (SDSQUA4-032G-GN6MT) and also tried SAMSUNG EVO 256GB (FAT32) (MB-ME256KA/AM)

I have tried re-flashing and re-formatting .img to both SDs and booting with data chord only,  power chord only, and data+power. I have used the instructions on Pwnagotchi site for Config.toml file and attempted boot without toml file. 

Hopefully someone will figure this one out, but I also will be trying to follow setup from here: https://www.youtube.com/watch?v=gyKT_mASSuc&t=609s and https://github.com/Xyl0se/Pwnagotchi-new-guerilla-guide as these seem to be the most recent instructions out there.

MCKLtech, 2023-11-23:
Found this by Google, I'm having this issue too. Has anyone solved it?

I've tried multiple SD cards and flashes.