evilsocket/pwnagotchi #1118: [On HOLD] Update code for tensorflow 2.x

Ra2-IFV, 2022-09-24:
**Is your feature request related to a problem? Please describe.**
[StackOverFlow](https://stackoverflow.com/questions/55082483/why-can-i-not-import-tensorflow-contrib-i-get-an-error-of-no-module-named-tenso)

**Describe the solution you'd like**
"tf.contrib no longer exist in Tensorflow 2.0.x and it's modules were moved."

**Describe alternatives you've considered**
Update codes

**Additional context**
tensorflow 1.x works fine, and this may take a long time, so I think we should not focus on this now.

llamasoft, 2022-10-04:
Unfortunately, upgrading to Tensorflow 2 may not be possible for a while. 🙁 

Tensorflow 2 _requires_ a 64-bit system but the Raspberry Pi Zero W is 32-bit.  The Zero 2 supports 64-bit but the system image pwnagotchi uses is 32-bit.  
Adding support for Tensorflow 2 would likely mean dropping support for the original RasPi Zero W.  Given that there's a massive Zero W 2 shortage, it's going to be a while before dropping Zero W 1 support is even reasonable.

Ra2-IFV, 2024-09-15:
Project archived.

closed 2024-09-15 by Ra2-IFV