evilsocket/pwnagotchi #1130: Waveshare V3 (B) support

timmosaurusrex, 2022-11-27:
Hi I have accidentally bought a Waveshare V3 color display. I was hoping that someone with some extra time can make a working driver with partial refresh. Sorry if this is a lot to ask. I tried making my own and it kinda worked, only without partial refresh and after is refreshes my display turns red. Thanks for your time.

user171, 2023-01-02:
Please forgive my ignorance, but where do you find that driver?

srepollock, 2023-02-18:
I found some success when setting `ui.display.type` to `waveshare213d`; but also noted that `waveshare213b` is an option

ch0en3, 2023-10-14:
> I found some success when setting `ui.display.type` to `waveshare213d`; but also noted that `waveshare213b` is an option

Can you tell me which image version you are using? 1.55 or 1.56 beta? I found this article but following the above does not make it work successfully。https://pwnsec.io/how-to-make-a-pwnagotchi-in-2022-with-a-waveshare-v3-screen/