evilsocket/pwnagotchi #1132: pi zero 2W bettercap is blind with error message "wifi error while hopping to channel 14: iw: out= err=exit status 234"

aluminum-ice, 2022-12-25:
I have pwnagotchi mostly working on a pi zero 2W except that bettercap finds no AP.  When I check syslog, I see this error message over and over:

[sys.log] [war] wifi error while hopping to channel 14: iw: out= err=exit status 234

I have read through the bettercap repository issues and see this problem mentioned but no resolution. Anyone have any ideas?

makerph, 2023-01-13:
encounter this also.

```
Jan 12 23:52:34 pot bettercap-launcher[456]: [23:52:34] [sys.log] [war] wifi error while hopping to channel 12: iw: out= err=exit status 234
Jan 12 23:52:35 pot bettercap-launcher[456]: [23:52:35] [sys.log] [war] wifi error while hopping to channel 13: iw: out= err=exit status 234
Jan 12 23:52:35 pot bettercap-launcher[456]: [23:52:35] [sys.log] [war] wifi error while hopping to channel 14: iw: out= err=exit status 234
Jan 12 23:52:38 pot bettercap-launcher[456]: [23:52:38] [sys.log] [war] wifi error while hopping to channel 12: iw: out= err=exit status 234
Jan 12 23:52:39 pot bettercap-launcher[456]: [23:52:39] [sys.log] [war] wifi error while hopping to channel 13: iw: out= err=exit status 234
Jan 12 23:52:39 pot bettercap-launcher[456]: [23:52:39] [sys.log] [war] wifi error while hopping to channel 14: iw: out= err=exit status 234
Jan 12 23:52:42 pot bettercap-launcher[456]: [23:52:42] [sys.log] [war] wifi error while hopping to channel 12: iw: out= err=exit status 234
Jan 12 23:52:43 pot bettercap-launcher[456]: [23:52:43] [sys.log] [war] wifi error while hopping to channel 13: iw: out= err=exit status 234
Jan 12 23:52:43 pot bettercap-launcher[456]: [23:52:43] [sys.log] [war] wifi error while hopping to channel 14: iw: out= err=exit status 234
Jan 12 23:52:46 pot bettercap-launcher[456]: [23:52:46] [sys.log] [war] wifi error while hopping to channel 12: iw: out= err=exit status 234
Jan 12 23:52:47 pot bettercap-launcher[456]: [23:52:47] [sys.log] [war] wifi error while hopping to channel 13: iw: out= err=exit status 234
Jan 12 23:52:47 pot bettercap-launcher[456]: [23:52:47] [sys.log] [war] wifi error while hopping to channel 14: iw: out= err=exit status 234
Jan 12 23:52:50 pot bettercap-launcher[456]: [23:52:50] [sys.log] [war] wifi error while hopping to channel 12: iw: out= err=exit status 234
Jan 12 23:52:51 pot bettercap-launcher[456]: [23:52:51] [sys.log] [war] wifi error while hopping to channel 13: iw: out= err=exit status 234
Jan 12 23:52:51 pot bettercap-launcher[456]: [23:52:51] [sys.log] [war] wifi error while hopping to channel 14: iw: out= err=exit status 234
```

makerph, 2023-01-13:
I've try editing
etc/default/crda
and set to country that supports channel 12-13
then check :
iwlist channel
channels 12-14 is  now there. previously not. then reboot pi, the error was gone and replace by this error.
```
Jan 13 12:36:22 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
Jan 13 12:36:22 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
Jan 13 12:36:23 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
Jan 13 12:36:26 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
Jan 13 12:36:26 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
Jan 13 12:36:27 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
Jan 13 12:36:30 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
Jan 13 12:36:30 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
Jan 13 12:36:31 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
Jan 13 12:36:34 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
Jan 13 12:36:34 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
Jan 13 12:36:35 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
Jan 13 12:36:38 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
Jan 13 12:36:38 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
Jan 13 12:36:39 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
Jan 13 12:36:42 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4108, -52
Jan 13 12:36:42 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -52
Jan 13 12:36:43 pot kernel: brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4110, -52
```

makerph, 2023-01-13:
I manually set the channel to recon to:  

personality.channels = [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]

excluding channels 12, 13 and 14. the error seems gone now. I will monitor is bettercap to see if it sees any AP.

DEDpeCK, 2024-09-01:
> Я отменил свои изменения в etc / default / crda и вручную установил канал в recon на:
> 
> personality.channels = [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]
> 
> исключая каналы 12, 13 и 14. похоже, ошибка исчезла. Я буду мониторить is bettercap, чтобы узнать, видит ли он какую-либо точку доступа.

Could you tell me please, in which file you manually set the channel?