evilsocket/pwnagotchi #1137: [BUG] preview.py script throws an AttributeError

therocco, 2023-01-21:
**Describe the bug**
While trying to run the `scripts/preview.py` script to locally test some template modifications I was making for myself I discovered it was erroring out due to an `AttributeError: NoneType` in `pwnagotchi/ui/fonts.py` on line 27

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate to the scripts folder in your terminal
2. run `./preview.py --displays waveshare_v2`

**Expected behavior**
A clear and concise description of what you expected to happen.

A `preview.png` image should be generated

**Screenshots**
If applicable, add screenshots to help explain your problem.

```
Traceback (most recent call last):
  File "preview.py", line 214, in <module>
    SystemExit(main())
  File "preview.py", line 144, in main
    display = CustomDisplay(config=config, state={'name': f"{display_type}>"})
  File "preview.py", line 21, in __init__
    super(CustomDisplay, self).__init__(config, state)
  File "/home/rocco/dev/pwnagotchi/scripts/../pwnagotchi/ui/display.py", line 12, in __init__
    super(Display, self).__init__(config, hw.display_for(config), state)
  File "/home/rocco/dev/pwnagotchi/scripts/../pwnagotchi/ui/hw/__init__.py", line 44, in display_for
    return WaveshareV2(config)
  File "/home/rocco/dev/pwnagotchi/scripts/../pwnagotchi/ui/hw/waveshare2.py", line 9, in __init__
    super(WaveshareV2, self).__init__(config, 'waveshare_2')
  File "/home/rocco/dev/pwnagotchi/scripts/../pwnagotchi/ui/hw/base.py", line 25, in __init__
    'font': fonts.status_font(fonts.Medium),
  File "/home/rocco/dev/pwnagotchi/scripts/../pwnagotchi/ui/fonts.py", line 27, in status_font
    return ImageFont.truetype(STATUS_FONT_NAME, size=old_font.size + SIZE_OFFSET)
AttributeError: 'NoneType' object has no attribute 'size'
```

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.5.5
 - OS version: Raspbian and Ubuntu 22.04
 - Type of hardware: Laptop, desktop, and Raspberry Pi Zero W

D337z, 2023-04-15:
The preview.py script is just a screen demo from Waveshare which can be compiled from their website.  It demonstrates what the screen can be used for and has no bearing what-so-ever on this project other than being included with the screen's configuration files.

therocco, 2023-04-15:
> The preview.py script is just a screen demo from Waveshare which can be compiled from their website. It demonstrates what the screen can be used for and has no bearing what-so-ever on this project other than being included with the screen's configuration files.

Thank you for your feedback but I would respectfully disagree. It is a script I used for testing new screens/themes that was slowing me down due to the bug when I was working on multiple new screens back in January. Since it was included with the project and others might encounter the bug it seemed prudent to create this issue for it as well as a pull request located at https://github.com/evilsocket/pwnagotchi/pull/1138 that provides a fix. 

This way the bug itself is detailed as much as possible and the fix is detailed as much as possible so it hopefully provides @evilsocket a little piece of mind and context when they QA and validate the fix if it is deemed fit for merging. That's the beauty of open source. If you see a bug you can squash a bug and hopefully someone somewhere finds it useful. :+1: